lwfinger/rtw8852be #3: worked fine for me on pop os. but couldn't make it work manjaro

saikatm, 2022-06-23:
how to install it on manjaro? can you help please

saikatm, 2022-06-23:
sir?

lwfinger, 2022-06-23:
Let me explain how time zones work! Your first request came in after I went to bed, and your second before I got up. I get really upset when I get pings for an issue that I have not even seen!!!!

Remember I do not work for you, or anyone else. I am a volunteer.

I have no idea what needs to be done on Manjaro. I do not understand people that change distros constantly. I have used openSUSE, and its predecessor SuSE, for 20 years. Whenever I need to install another version because some idiot cannot make it work, the experience makes me glad I use openSUSE.

https://linuxconfig.org/manjaro-linux-kernel-headers-installation will show you how to install kernel headers. Remember, google is your friend.

closed 2022-06-23 by lwfinger

saikatm, 2022-06-23:
well thanks and sorry. thanks for the brief explanation and sorry for being rude. I am a linux newbie I don't have much knowledge about linux... I will try to make it install on a arch based distro...

saikatm, 2022-06-23:
btw do you work for realtek ?? how did you made the driver :open_mouth: and thanks a lot for making it. searched a lot fr the driver as I just got my new laptop

lwfinger, 2022-06-23:
No, I do not work for Realtek. I am an unpaid volunteer that attempts to provide drivers for two situations: 1. I take the new drivers from the kernel and backport them to build on older kernels, and 2. I acquire drivers from various sources and make them to build on the latest kernels, and fix the problems with the drivers. I do have a good working relationship with the lead programmer on Realtek's PCI development team.

This driver was sent to me by a user of a new laptop with an rtw8892be chip. He got the driver from HP.

saikatm, 2022-08-05:
how do you do all this complex programming stuff? can I follow you on social sites?

lwfinger, 2022-08-05:
I started working on kernel stuff during the kernel transition from version 2.4.x to 2.6.x. I knew very little about C programming in those days, but I read and learned. I have always had good programming skills from the time I started coding in 1964.

I do not do social media. I belong to Facebook, but I never post there - only read what friends and family post.

OnlineArts, 2022-08-05:
@lwfinger 
Thank you very much for providing the driver. Without your work I could not do my research projects on my new Lenovo notebook!

saikatm, 2022-08-05:
> I started working on kernel stuff during the kernel transition from version 2.4.x to 2.6.x. I knew very little about C programming in those days, but I read and learned. I have always had good [...]

that's really cool man. seems like you are an old guy. I am really very curious about how you manage and code all the drivers amd fix bugs.. I wish I could have a chat with you.

saikatm, 2022-08-05:
> @lwfinger Thank you very much for providing the driver. Without your work I could not do my research projects on my new Lenovo notebook!
> 
> By the way, the driver works fine with Ubuntu 22.04.

same here. I bought the new hp laptop and installed linux but found the wifi doesn't work. did googling for 2 days and tried wrong drivers but luckily found this repo.