lwfinger/rtw8852be #4: WiFi only?

alexleosumy, 2022-06-29:
Hello, I have Laptop redmi notebook 14 pro. I installed your driver for WiFi Thanks!!!
As I  understand this chip is a combo WiFi + Bluetooth https://wikidevi.wi-cat.ru/Realtek_RTL8852BE_Combo_Module 
Does your driver support Bluetooth?
During boot, I cannot see any error
```
dmesg | grep Bluetooth                                                                                                                 
[    1.536391] usb 1-4: Product: Bluetooth Radio
[    3.120330] Bluetooth: Core ver 2.22
[    3.120396] Bluetooth: HCI device and connection manager initialized
[    3.120404] Bluetooth: HCI socket layer initialized
[    3.120408] Bluetooth: L2CAP socket layer initialized
[    3.120416] Bluetooth: SCO socket layer initialized
[    3.351544] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.351552] Bluetooth: BNEP filters: protocol multicast
[    3.351561] Bluetooth: BNEP socket layer initialized
[   11.383222] Bluetooth: RFCOMM TTY layer initialized
[   11.383236] Bluetooth: RFCOMM socket layer initialized
[   11.383250] Bluetooth: RFCOMM ver 1.11
```
I can get info about Bluetooth using bluetoothctl:
```
bluetoothctl                                                                                                                               
Agent registered
[bluetooth]# show
Controller 34:6F:24:D3:98:D1 (public)
	Name: LMain
	Alias: LMain
	Class: 0x007c010c
	Powered: yes
	Discoverable: yes
	DiscoverableTimeout: 0x00000000
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0540
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0a (10)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
```
But I cannot pair any device via console or GUI. Search always returns empty result.

I use Manjaro with kernel 5.15.49-1-MANJARO.
```
lsusb 
Bus 001 Device 003: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
```
On Windows OS all work correctly. I use drivers from China website

lwfinger, 2022-06-29:
This driver is for wifi only. The BT part uses drivers btusb and btrtl, which are part of the standard kernel; however, based on an issue in the rtw89 repo, I know that your device is NOT in the standard kernel. I will submit a patch, but I need to know the part of 'cat /sys/kernel/debug/usb/devices' that applies to the Bluetooth device.

lwfinger, 2022-06-30:
It is your lucky day! As I got two requests for addition of Opticis BT devices, I decided to create a new code repository for the modified BT codes.

In a console terminal, do the following:
git clone git@github.con:lwfinger/rtw89-BT.git
cd rtw89-bt
make
sudo make install

If you are using secure boot, replace that install step with
sudo make sign-install

If you need to do this, please read the appropriate section of README.md so that you do not end up making your computer unbootable.

Good luck.

saikatm, 2022-08-15:
can add the Bluetooth driver in the repo or in a different branch ??

lwfinger, 2022-08-15:
The BT driver is in my repo rtw89-BT. The BT and wifi devices are separate interfaces on the same chip.

saikatm, 2022-08-15:
Thanks just tried worked like a charm!

lwfinger, 2022-08-15:
Goog to hear.

closed 2022-08-15 by lwfinger

saikatm, 2022-08-15:
this repo has a Bluetooth branch. does it works?

lwfinger, 2022-08-15:
No. Do a git pull. After that README.md will show you how to get the BT driver.