lwfinger/rtw8852be #8: Issue with 5.18.6

mauricioquintela, 2022-08-06:
Running arch, updated kernel today to 5.18.6-arch1. 
Modules were working fine on kernel 5.18.5, prior to this update.

I have clean rebuild the module, with 
```
git pull
make clean
make
sudo make install
```
and wireless is still not working after reboot. No errors are outputed in either step of the build process. 
Also tested deleting the local repo and cloning again from the begining, to the same effect.

Any assistance would be appreciated. Laptop is a thinkpad E15 gen4.

lwfinger, 2022-08-06:
There should be no difference between 5.18.5 and 5.18.6.

What do you see when you enter the following two commands?

sudo modprobe -rv 8852be
sudo modprobe -v 8852be

If you get an error with the second one, look at the end of the dmesg output.

nestlecrunchkid, 2022-08-06:
Output of the second one:
insmod /lib/modules/5.18.16-200.fc36.x86_64/kernel/drivers/net/wireless/8852be.ko

mauricioquintela, 2022-08-06:
> There should be no difference between 5.18.5 and 5.18.6.
> 
> What do you see when you enter the following two commands?
> 
> sudo modprobe -rv 8852be sudo modprobe -v 8852be
> 
> If you get an error with the second one, look at the end of the dmesg output.

Yesterday I ended up getting it to work with https://github.com/HRex39/rtl8852be, following the 5.18 kernel guide. 
Before that, I ran `sudo modprobe -v 8852be`, there was no output but it also didn't error out. How would you propose I remove the other repo to keep testing this one?

I also tested the [aur package](https://aur.archlinux.org/packages/rtw8852be-dkms-git) for your repo and it was failing with
```
(...)
(2/2) Install DKMS modules
==> dkms install --no-depmod rtw8852be/r19.8c2ce89 -k 5.18.16-arch1-1
Error!  Build of rtw8852be.ko failed for: 5.18.16-arch1-1 (x86_64)
Make sure the name of the generated module is correct and at the root of the
build directory, or consult make.log in the build directory
/var/lib/dkms/rtw8852be/r19.8c2ce89/build for more information.
==> WARNING: `dkms install --no-depmod rtw8852be/r19.8c2ce89 -k 5.18.16-arch1-1' exited 7
```
which might be related to why the git repo was not working. No errors were present in the `make.log` log file.

lwfinger, 2022-08-06:
I did not write the dkms config file, and cannot support it. Perhaps someone else will see your report, and help you with it.

The only problem with having multiple sources of a driver from multiple repos is that you have no idea of which one is installed, or if they makefile gives them different names, then you will end up with competing drivers getting in each other's way with neither of them working.

mauricioquintela, 2022-08-06:
This issue is not regarding the dkms package, though. I merely posted its error in case it was useful. 

My issue is with the version provided by this repository, which does not provide any error when building  and installing, but afterwards behaves as if I have not installed it.

lwfinger, 2022-08-06:
Without the -v option, the modprobe command will only produce output if the module cannot be loaded. With the -v option, it will not produce any output is the module is already loaded.

If that other repo works for you, use it. I looked at the differences between that repo and mine, and I have no idea where those changes came from. The author gives no hints in the commit message. I repeat - use it if it works.

closed 2022-08-06 by lwfinger