lwfinger/rtw8852be #9: Issue with 5.18.6 kernel

nestlecrunchkid, 2022-08-07:
Doesn't work with 5.18.6 kernel. Hrex39's repo doesn't work either

lwfinger, 2022-08-08:
OK. Please do the following:

sudo modprobe -rv 8852be
sudo modprobe -v 8852be
dmesg > dmesg.txt

Post the output of the first two, and attach file dmesg.txt to this issue.

nestlecrunchkid, 2022-08-08:
How do I get access to dmesg.txt?

nestlecrunchkid, 2022-08-08:
https://user-images.githubusercontent.com/81821226/183390393-a8b959d2-53d7-4efa-b9af-91a3ab402b23.mp4

lwfinger, 2022-08-08:
I gave you the command on how to create dmesg.txt. It is possible that you need to use sudo in front of it. Some idiot decided that  the dmesg output contained sensitive info and set the system to require root privilege. Yours may be one of them.

BTW, Please stop attaching videos. Instead copy and paste any output into the issue window. I could barely see the video on my laptop screen. I did see enough to see that the build, install, and module load are working. I do need the dmesg output.

JulienDamers, 2022-08-08:
Hi there, I think I might have the same problem. Here is the dmesg.txt file after unloading and loading the module

[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9284696/dmesg.txt)

Also, here is the output of the *lspci -knn* command for the network controller
 
`
03:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
	Subsystem: Lenovo Device [17aa:4853]
	Kernel modules: 8852be

`

Phosphenius, 2022-08-08:
Hijacking this 'cause I too have an issue with `5.18.16`. It used to work fine except for occasional crashes where I lost connection (apparently some issue with a memory overrun). However, with the most recent changes, I can build the module but when I `modprobe` it, nothing happens. No device shows up in the `ip l` output. I've `git bisect`ed this to be caused by this commit: 8551ffb1b262f0d74b835eeebfdbe747c60a0ea0

From the looks of it this is supposed to fix the other issue (mentioned above) I've had with a less recent version. I was on 62e8be6b47522a2471840bbfdad8cf9d33fff7d7 IIRC.

```shell
uname -r
5.18.16-arch1-1

sudo modprobe -rv 8852be
rmmod 8852be
rmmod cfg80211

sudo modprobe -v 8852be 
insmod /lib/modules/5.18.16-arch1-1/kernel/net/wireless/cfg80211.ko.zst 
insmod /lib/modules/5.18.16-arch1-1/kernel/drivers/net/wireless/8852be.ko 
```

[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9284724/dmesg.txt)

nestlecrunchkid, 2022-08-09:
[dmesgmodprobe-rv.txt](https://github.com/lwfinger/rtw8852be/files/9286788/dmesgmodprobe-rv.txt)
[dmesgmodprobe-v.txt](https://github.com/lwfinger/rtw8852be/files/9286789/dmesgmodprobe-v.txt)

HRex39, 2022-08-10:
> Hijacking this 'cause I too have an issue with `5.18.16`. It used to work fine except for occasional crashes where I lost connection (apparently some issue with a memory overrun). However, with the most recent changes, I can build the module but when I `modprobe` it, nothing happens. No device shows up in the `ip l` output. I've `git bisect`ed this to be caused by this commit: [8551ffb](https://github.com/lwfinger/rtw8852be/commit/8551ffb1b262f0d74b835eeebfdbe747c60a0ea0)
> 
> From the looks of it this is supposed to fix the other issue (mentioned above) I've had with a less recent version. I was on [62e8be6](https://github.com/lwfinger/rtw8852be/commit/62e8be6b47522a2471840bbfdad8cf9d33fff7d7) IIRC.
> 
> ```shell
> uname -r
> 5.18.16-arch1-1
> 
> sudo modprobe -rv 8852be
> rmmod 8852be
> rmmod cfg80211
> 
> sudo modprobe -v 8852be 
> insmod /lib/modules/5.18.16-arch1-1/kernel/net/wireless/cfg80211.ko.zst 
> insmod /lib/modules/5.18.16-arch1-1/kernel/drivers/net/wireless/8852be.ko 
> ```
> 
> [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9284724/dmesg.txt)

> Hijacking this 'cause I too have an issue with `5.18.16`. It used to work fine except for occasional crashes where I lost connection (apparently some issue with a memory overrun). However, with the most recent changes, I can build the module but when I `modprobe` it, nothing happens. No device shows up in the `ip l` output. I've `git bisect`ed this to be caused by this commit: [8551ffb](https://github.com/lwfinger/rtw8852be/commit/8551ffb1b262f0d74b835eeebfdbe747c60a0ea0)
> 
> From the looks of it this is supposed to fix the other issue (mentioned above) I've had with a less recent version. I was on [62e8be6](https://github.com/lwfinger/rtw8852be/commit/62e8be6b47522a2471840bbfdad8cf9d33fff7d7) IIRC.
> 
> ```shell
> uname -r
> 5.18.16-arch1-1
> 
> sudo modprobe -rv 8852be
> rmmod 8852be
> rmmod cfg80211
> 
> sudo modprobe -v 8852be 
> insmod /lib/modules/5.18.16-arch1-1/kernel/net/wireless/cfg80211.ko.zst 
> insmod /lib/modules/5.18.16-arch1-1/kernel/drivers/net/wireless/8852be.ko 
> ```
> 
> [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9284724/dmesg.txt)

What do you think the problem with modprobe not showing wifi icon? How do you want to solve it?

HRex39, 2022-08-10:
In some dmesg.txt mentioned above, I saw some message called:
```
8852be: loading out-of-tree module taints kernel.
8852be: module verification failed: signature and/or required key missing - tainting kernel
```

Does this affect the normal startup of the driver?

lwfinger, 2022-08-10:
No, tainting the kernel is not a problem.

The problem is that new kernels issue a warning when you try to vmalloc a 0-length piece. I am trying to find the correct fixup when the condition occurs. The main problem is that every failure requires a reboot, and I am in the middle of a high priority project where the builds take 45 minutes, thus my opportunities to reboot are small. I am working on the problem, but slowly.

erotallgeese, 2022-08-11:
Hello,
I meet same issue that it get crash at rtw_efuse_init and i found it may caused by [this patch](https://github.com/lwfinger/rtw8852be/commit/8551ffb1b262f0d74b835eeebfdbe747c60a0ea0)

I thought sizeof(void *) would always be 8.
After revert this, i can then insmod normally.

lwfinger, 2022-08-11:
I fixed the problem and pushed the solution. Sorry for the delay as I had a high priority job that occupied about 2 weeks.

On a really recent kernel (6.0-rc0), there is one warning logged, but it does not cause any harm.

closed 2022-08-11 by lwfinger

nestlecrunchkid, 2022-08-13:
It works now. Thanks!