lwfinger/rtw8852be #11: Driver not working 5.15.0-43-generic (secure boot)

bashanov, 2022-08-08:
Driver had been working perfectly for one month and recently wifi icon disappeared.
After reinstall it's still not working.

```
:~$ inxi -nn
Network:
  Device-1: Realtek driver: N/A
  IF-ID-1: br-b3108ebf6c7c state: up speed: 10000 Mbps duplex: unknown
    mac: 02:42:7c:2e:26:4a
```

```
:~$ lspci -nn
01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
```

After I did disable/enable driver via modprobe, got the following dmesg output (in attachment):
[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9281083/dmesg.txt)

lwfinger, 2022-08-11:
I added code to sign the driver. Use 'sudo make sign-install' rather than 'sudo make install'. You will need to reboot, but be careful when the MOK bluescreen comes up on reboot.

bashanov, 2022-08-11:
I saw. I did only sign installs. I sign, reboot and enroll the key, but driver still not works.

lwfinger, 2022-08-11:
Did you do a new pull? I just pushed a bug fix.

bashanov, 2022-08-11:
After pull wifi works again! Thanks!

closed 2022-08-11 by lwfinger