lwfinger/rtw8852be #12: driver stopped working on 5.15.59-1 kernel

saikatm, 2022-08-12:
Just updated to 5.15.59-1-MANJARO kernel. 
compiled and installed and rebooted did not work. 

- cd rtw8852be
- make clean
-  make 
- sudo make install
- sudo modprobe -v 8852be
- reboot

saikatm, 2022-08-12:
just did git pull for the latest commit and recompiled it. WiFi is back!

closed 2022-08-12 by saikatm

lwfinger, 2022-08-12:
Always do a git pull. You never know what changes might be there.

saikatm, 2022-08-12:
> Always do a git pull. You never know what changes might be there.

thanks man. I don't want the frequent kernel updates on manjaro to break my wifi...

saikatm, 2022-08-12:
would u sugget using this? https://aur.archlinux.org/packages/rtw8852be-dkms-git OR should I stick with the repo

lwfinger, 2022-08-12:
It is up to you. If the arch dkms works, then you do not have to build it yourself. As far as I know, their sources for the driver come from this repo.