lwfinger/rtw8852be #15: "Realtek Semiconductor Co., Ltd. Device b852" driver problem

jixiedaima, 2022-08-21:
Installed this driver on ubuntu22.04, but the computer still can't recognize wifi.
Could you tell me how to install the driver?

thinkbook@thinkbook:/Downloads/test/RTL88x2BU-Linux-Driver$ lspci | grep Network
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
thinkbook@thinkbook:$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 04f2:b758 Chicony Electronics Co., Ltd Integrated Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 005: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 004 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 005: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 002: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lwfinger, 2022-08-21:
This is a PCI device. To see it in the scan, use the command 'lspci -nn'. The lsusb scan does show the Bluetooth device. I will make sure that your device is in the tables for  https://github.com/lwfinger/rtw89-BT.git, which is where the updates BT drivers are kept.

jixiedaima, 2022-08-22:
Hi, here is my info:
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b5] (rev 01)
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b6]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b7] (rev 01)
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b8]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b7] (rev 01)
00:02.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14ba]
00:02.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14ba]
00:02.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14ba]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b7] (rev 01)
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b7] (rev 01)
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b7] (rev 01)
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b9] (rev 10)
00:08.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14b9] (rev 10)
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 71)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1679]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167a]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167b]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167c]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167d]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167e]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:167f]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:1680]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA107BM [GeForce RTX 3050 Ti Mobile] [10de:25e0] (rev a1)
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
03:00.0 SD Host controller [0805]: O2 Micro, Inc. SD/MMC Card Reader Controller [1217:8621] (rev 01)
04:00.0 Non-Volatile memory controller [0108]: Phison Electronics Corporation PS5013 E13 NVMe Controller [1987:5013] (rev 01)
05:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [1002:1681] (rev 02)
05:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:1640]
05:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] VanGogh PSP/CCP [1022:1649]
05:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:161d]
05:00.4 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:161e]
05:00.5 Multimedia controller [0480]: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor [1022:15e2] (rev 60)
05:00.6 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller [1022:15e3]
06:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:161f]
06:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d6]
06:00.4 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d7]

lwfinger, 2022-08-22:
That device is in the tables, and should be loaded. Please do the following:

lsmod | grep 88
sudo modprobe -rv 8852be
sudo modprobe -v 8852be

Post the output of all 3. If the 3rd command errors, then look at the end of the dmesg output and post the error message. I do not need, or want, the entire dmesg output.

jixiedaima, 2022-08-23:
hinkbook@thinkbook:~$ lsmod | grep 88
nls_iso8859_1          16384  1
ipmi_msghandler       122880  1 ipmi_devintf
r8152                 114688  1 r8153_ecm
thinkbook@thinkbook:~$ sudo modprobe -rv 8852be
[sudo] thinkbook 的密码： 
modprobe: FATAL: Module 8852be not found.
thinkbook@thinkbook:~$ sudo modprobe -v 8852be
modprobe: FATAL: Module 8852be not found in directory /lib/modules/5.15.0-47-generic

    3.920054] ucsi_acpi USBC000:00: PPM init failed (-110)
[    9.344301] kauditd_printk_skb: 33 callbacks suppressed
[    9.344304] audit: type=1400 audit(1661290447.415:44): apparmor="DENIED" operation="capable" profile="/usr/sbin/cups-browsed" pid=1169 comm="cups-browsed" capability=23  capname="sys_nice"
[   13.767621] audit: type=1400 audit(1661290451.835:45): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1204 comm="snap-confine" capability=12  capname="net_admin"
[   13.767636] audit: type=1400 audit(1661290451.835:46): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1204 comm="snap-confine" capability=38  capname="perfmon"
[   13.777498] audit: type=1400 audit(1661290451.847:47): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1204 comm="snap-confine" capability=4  capname="fsetid"
[   14.076381] Bluetooth: RFCOMM TTY layer initialized
[   14.076397] Bluetooth: RFCOMM socket layer initialized
[   14.076404] Bluetooth: RFCOMM ver 1.11
[   14.787530] audit: type=1400 audit(1661290452.855:48): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1473 comm="snap-confine" capability=12  capname="net_admin"
[   14.787538] audit: type=1400 audit(1661290452.855:49): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=1473 comm="snap-confine" capability=38  capname="perfmon"
[   14.859240] audit: type=1326 audit(1661290452.927:50): auid=1000 uid=1000 gid=1000 ses=2 subj=? pid=1473 comm="snapd-desktop-i" exe="/snap/snapd-desktop-integration/14/bin/snapd-desktop-integration" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f83f7f6373d code=0x50000
[   15.300573] rfkill: input handler disabled
[   30.961171] IPv6: ADDRCONF(NETDEV_CHANGE): enx00e04c8042bf: link becomes ready
[   30.961489] r8152 4-2.4:1.0 enx00e04c8042bf: carrier on
[   42.415650] audit: type=1400 audit(1661290480.483:51): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=2185 comm="snap-confine" capability=12  capname="net_admin"
[   42.415660] audit: type=1400 audit(1661290480.483:52): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=2185 comm="snap-confine" capability=38  capname="perfmon"
[   42.419006] audit: type=1400 audit(1661290480.487:53): apparmor="DENIED" operation="capable" profile="/usr/lib/snapd/snap-confine" pid=2185 comm="snap-confine" capability=4  capname="fsetid"
[   42.467061] audit: type=1400 audit(1661290480.535:54): apparmor="DENIED" operation="mkdir" profile="snap-update-ns.firefox" name="/usr/share/libreoffice/help/" pid=2207 comm="5" requested_mask="c" denied_mask="c" fsuid=0 ouid=0
[   42.467368] audit: type=1400 audit(1661290480.535:55): apparmor="DENIED" operation="open" profile="snap-update-ns.firefox" name="/var/lib/" pid=2207 comm="5" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   43.011205] audit: type=1326 audit(1661290481.079:56): auid=1000 uid=1000 gid=1000 ses=2 subj=? pid=2185 comm="firefox" exe="/snap/firefox/1635/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7fd8c70f073d code=0x50000
[   43.422591] audit: type=1326 audit(1661290481.491:57): auid=1000 uid=1000 gid=1000 ses=2 subj=? pid=2185 comm="Renderer" exe="/snap/firefox/1635/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=312 compat=0 ip=0x7fd8c70f073d code=0x50000
[   43.479019] audit: type=1326 audit(1661290481.547:58): auid=1000 uid=1000 gid=1000 ses=2 subj=? pid=2185 comm="firefox" exe="/snap/firefox/1635/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=312 compat=0 ip=0x7fd8c70f073d code=0x50000
[   43.479026] audit: type=1326 audit(1661290481.547:59): auid=1000 uid=1000 gid=1000 ses=2 subj=? pid=2185 comm="firefox" exe="/snap/firefox/1635/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=312 compat=0 ip=0x7fd8c70f073d code=0x50000

lwfinger, 2022-08-25:
Pull again, cd HRex39, make, and sudo make install. After a reboot, you should have wifi.

There is one problem with this driver that I have not yet fixed. Under heavy load, such as running speedtest.net, the driver runs out of DMA buffers and locks up. When this happens, only a reboot will revover from the condition.

jixiedaima, 2022-08-25:
Now, I can see wifi. thank you! If the problem solved, please let me know. thank you!

lwfinger, 2022-08-29:
Do another pull. I have fixed the DMA problem.

closed 2022-08-29 by lwfinger