lwfinger/rtw8852be #16: Not working on 5.18.18-200 on Fedora 36

ljcoogan, 2022-08-22:
This morning I updated the kernel on my Fedora installation to 5.18.18-200. After, I reinstalled the driver by deleting the directory where I had cloned this repository and cloning it anew, before running `make` and `sudo make sign-install`, the output of which is attached. 

Before I did this, the drivers worked fine. Now, the WiFi has stopped working. Here is the output of some commands:

**uname -r:**
`5.18.18-200.fc36.x86_64`

**modprobe -v 8852be:**
`modprobe: FATAL: Module 8852be not found in directory /lib/modules/5.18.18-200.fc36.x86_64`

**lspci -nn | grep Realtek**
`01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]`

Find the output of dmesg attached as well.

Thanks for any help you can offer

[sign-install.txt](https://github.com/lwfinger/rtw8852be/files/9393410/sign-install.txt)
[make.txt](https://github.com/lwfinger/rtw8852be/files/9393411/make.txt)
[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9393412/dmesg.txt)

ljcoogan, 2022-08-22:
I don't believe the problem lies with secure boot - I just tried to fix this by resetting secure boot to its factory state and reinstalling the driver with its key. WiFi is still not working.

lwfinger, 2022-08-22:
I did not see anything wrong in your fixes, but I did see a problem with the Makefile. Do a 'git pull' and repeat the 'sudo make sign install'.

ljcoogan, 2022-08-22:
Now I'm greeted by a new error when I run sudo make sign-install:

```
At main.c:291:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: 8852be.ko: No such file or directory
make: *** [Makefile:99: sign] Error 1
```

lwfinger, 2022-08-22:
Yes, I just found that one too. Do another pull.

ljcoogan, 2022-08-22:
I'm still getting the same error unfortunately - after pulling didn't change anything, I just deleted and cloned again the repo and it still hasn't fixed it.

lwfinger, 2022-08-22:
Somehow the repo got mixed up with the rtw90 repo, which is for rtw8852a and rtw8852c.

Wipe out all your local repo and reclone rtw8852be. It works for me. :)

ljcoogan, 2022-08-22:
I deleted the local repo, reset my secure boot keys to factory state, and proceeded to reclone and install the driver. The install proceeded with no errors, I rebooted and went through the MOK process... and I have no WiFi.

However, there is an 8852be module installed. After unloading the module, running **sudo modprobe -v 8852be** returns:
```
insmod /lib/modules/5.18.18-200.fc36.x86_64/kernel/net/wireless/cfg80211.ko.xz
insmod /lib/modules/5.18.18-200.fc36.x86_64/kernel/drivers/net/wireless/8852be.ko
```
Unfortunately, the module doesn't seem to do anything, or at any rate, doesn't provide me WiFi. I have attached an updated dmesg.txt, which does mention 8852be, and does not appear to show an error message, although to be honest I wouldn't know how to spot one if it did.

Thanks for all your help so far, sorry this is taking so long

[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9397443/dmesg.txt)

ljcoogan, 2022-08-23:
Unfortunately, the patch you made a few hours ago to fix installs on Fedora has not done the trick on my machine.

While installing, I noticed towards the end of the output of sign make-install it says this:
`install -p -m 644 8852be.ko  /lib/modules/5.18.18-200.fc36.x86_64/kernel/drivers/net/wireless//realtek/rtw89/`
The double slash between "wireless" and "realtek" isn't the problem here, is it?

ljcoogan, 2022-08-23:
I just modified my copy of the makefile to fix the double forward slashes, i.e. removing the '/' after '$(MODDESTDIR)' so it reads:
```
install:
	install -p -m 644 $(MODULE_NAME).ko  $(MODDESTDIR)realtek/rtw89/
	/sbin/depmod -a ${KVER}
```
I then uninstalled, cleared secure boot keys, and reinstalled with the updated makefile

This time, sign make-install outputted:
` install -p -m 644 8852be.ko /lib/modules/5.18.18-200.fc36.x86_64/kernel/drivers/net/wireless/realtek/rtw89/`

WiFi still isn't working, but hey at least something was accomplished :)

lwfinger, 2022-08-23:
I made that change on my system, and pushed the changes.

I updated/upgraded my Fedora 35 system to Fedora 36. As you find, the driver is not recognized by the network system. I do not know how to fix this. It is not inherently due to Gnome/Wayland. I added that desktop to my openSUSE system, and the device was recognized there.

I tried switching  Fedora to use wicked rather that NetworkManager, but that broke networking on all devices.

I also tried installing the GitHub version of the driver for the RTL8188EU. Although the code is very similar, the device was recognized by Fedora.

I have no idea what else to try.

ljcoogan, 2022-08-23:
The problem only appeared for me after upgrading to kernel 5.18.18. I just logged in to 5.18.17 (Fedora keeps a few old kernels that you can access from GRUB) and WiFi is indeed working perfectly, I'm writing this on my Fedora system right now.

That means the .18 upgrade is a good place to start. The [Fedora Updates System](https://bodhi.fedoraproject.org/updates/FEDORA-2022-c782864bc6) page for this kernel has nobody reporting issues with WiFi, but I can try leaving a comment there if you think it's an issue on their end. The .18 [release notes](https://lwn.net/Articles/904958/) from Greg Kroah-Hartman lists nothing about the 8852be (presumably because it's not in the kernel), but does mention the following change for 88852a
```
Ping-Ke Shih (1):
      wifi: rtw89: 8852a: rfk: fix div 0 exception 
```

Aside from that, I unfortunately don't know where to look, I'm not very knowledgable about this topic.

ljcoogan, 2022-08-23:
I just upgraded to the [5.18.19 kernel](https://bodhi.fedoraproject.org/updates/FEDORA-2022-5674f93546) and installed the drivers, and it does not fix the problem.

lwfinger, 2022-08-25:
I installed the GitHub driver posted by HRex39 in the HRex39 directory of my repo. It does work with Fedora 5,18,18-100 on my system.

In the future, I will be examining the differences between my version  and this one. It is important as the new one runs out of DMA buffers and locks up the system if you push it hard, such as running speedtest.net. For that reason, it is not acceptable, but it does give people a working wifi.

ljcoogan, 2022-08-25:
Driver works for me - I'm not sure if you want to mark this as closed or not considering you want to further examine the issue, so I'll leave that for you to decide.

Thanks for all you do, and all your work on this problem, you have saved mine and many other people's experience on Linux - this is actually my _third_ device where I have had to use your drivers

byronsit, 2022-08-28:
I fount that not hard push ,only little but a long time, can also lead to the DMA buffer error. And the newest version still have the problem.

lwfinger, 2022-08-28:
That problem has not been fixed.