lwfinger/rtw8852be #17: Not working on arch 5.19.3

mauricioquintela, 2022-08-23:
After update to kernel 5.19.3, wifi no longer works. Has been updated following guidelines, as per usual:
```
git pull 
make clean
make 
sudo make install
```
No errors were present during compilation.

Output of `sudo modprobe -rv 8852be`:
```
> sudo modprobe -rv 8852be
rmmod 8852be
rmmod cfg80211
```
Output of `sudo modprobe -v 8852be`:
```
> sudo modprobe -v 8852be
insmod /lib/modules/5.19.3-arch1-1/kernel/net/wireless/cfg80211.ko.zst 
insmod /lib/modules/5.19.3-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/8852be.ko 
```

dmesg output: [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9402543/dmesg.txt)

lwfinger, 2022-08-23:
Your problem is the following line in dmesg.txt:
[ 4877.041434] ******************** log_efuse_size is zero

I have seen this line in other's output. I have no idea what is causing it. I do not see it on my openSUSE system.

mauricioquintela, 2022-08-23:
Any suggestions on what I can do to further test this issue? I will also test the other [8852be driver](https://github.com/HRex39/rtl8852be) later today, to see if that one works.

mauricioquintela, 2022-08-23:
> Any suggestions on what I can do to further test this issue? I will also test the other [8852be driver](https://github.com/HRex39/rtl8852be) later today, to see if that one works.

Also, in continuation to this comment, everything was working fine prior to updating (5.19.2 arch). 
Also tested zen kernel, same issue is present[dmesg_zen.txt](https://github.com/lwfinger/rtw8852be/files/9406941/dmesg_zen.txt).

mauricioquintela, 2022-08-23:
Important update: Managed to get the [rtl8852be](https://github.com/HRex39/rtl8852be) driver working, errors were related to the changes mentioned in https://github.com/lwfinger/rtw8852be/commit/e282fb4089ba0b9e57e0f19a09bf06a753f7bc81. Including these changes in their [ioctl_cfg80211.c](https://github.com/HRex39/rtl8852be/blob/dev/os_dep/linux/ioctl_cfg80211.c) the driver works.
This might not be usefull, as their driver appears to be somewhat different.
 
Changes were namely regarding:
cfg80211_rtw_get_channel (line 6335+)
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
static int cfg80211_rtw_get_channel(struct wiphy *wiphy,
	struct wireless_dev *wdev,
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 2))
	u32 link_id,
#endif
	struct cfg80211_chan_def *chandef)
{
```
cfg80211_rtw_stop_ap (line 5385+)
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 2))
static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev, unsigned int link_id)
#else
static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev)
#endif
{
```
and wdev-> current_bss vs connected (line 10699+)
```
	#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 19, 0)) || defined(COMPAT_KERNEL_RELEASE)
	if (wdev->current_bss) {
	#else
	if (wdev->connected) {
	#endif

```

lwfinger, 2022-08-24:
I pulled the HRex39 code and fixed it to build on new kernels. I have been testing it with mixed results. It spams the logs a lot, and when you push it hard such as by running speedtest.net, it runs out of DMA buffers (swiotb entries full), and freezes. I will be looking at that version, which is found in the HRex39 directory of my repo. Do a 'git pull', 'cd HRex39', and the normal make, etc. If dmesg shows the buffer problem, you will need to reboot.

byronsit, 2022-08-25:
> I pulled the HRex39 code and fixed it to build on new kernels. I have been testing it with mixed results. It spams the logs a lot, and when you push it hard such as by running speedtest.net, it runs [...]

yes,  "swiotb entries full error" was happend in dmesg, and I need to reboot. 
anyway, it works for me.  
Thank you for your attention! Looking forward to the next commit to fix the bug.

mauricioquintela, 2022-08-25:
The HRex39 directory seems to be working here as well, with some dmesg spam. Haven't run into full DMA buffers yet. Thank you for your attention. Looking forward to the next few commits! (Also looking forward to when drivers for this card get merged into mainline, but I won't hold my breath yet :p )

mauricioquintela, 2022-08-27:
Update regarding 5.19.4: main repository is still not working, same issue as before. HRex39 code works ok.

lwfinger, 2022-08-27:
That is strange. I found the problem with mine and pushed the fix last Thursday. If you do a 'git pull', do you get any new material? That was the last push I did.

A driver for this chip is on the TODO list at Realtek, but the last I heard was not until November.

I know a little more about the DMA problem, but not yet how to fix it. The driver fails to throttle the rest of the system when a large number of TX requests are presented. RX is OK. That is why the upload test in speedtest.net fails.

mauricioquintela, 2022-08-27:
After upgrading the kernel, I followed the usual steps of running a git pull (it downloaded a bunch of material, latest was 46139110d417360f6996b4b1943a04c892c37985 at the time), rebooting, make clean and then installing normally. After a second reboot, no wifi card was detected, so I assumed the issue still remained. As such, I followed your suggestion from a bit above in this thread and just installed the HRex directory. As I was in a bit of a rush, I did not get a copy of dmesg logs. 

I'm now free to test more, so I will just do make unninstall on HRex's stuff, install from the main repository and I'll post the output from the two modprobes and the dmesg logs here in a few minutes. 

I'm glad to hear that it is at least being looked at from Realtek's side.

mauricioquintela, 2022-08-27:
Just unninstalled HRex's modules, building from main repository after another pull results in an error regarding implicit function declaration:
```
❯ make -j8                                                                                                                                                   ─╯
#rm -f .symvers.8852be
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.4-zen1-1-zen/build M=/home/mauricioquintela/Scripts/rtw8852be  modules
make[1]: Entering directory '/usr/lib/modules/5.19.4-zen1-1-zen/build'
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/platform/platform_linux_pc_pci.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/osdep_service.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/osdep_service_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_cfg.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/os_intfs.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/xmit_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/mlme_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/recv_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/wifi_regd.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_android.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_proc.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/nlrtw.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/pci_intf.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/pci_ops_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_mp.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_efuse.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/core/rtw_cmd.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/core/rtw_security.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/core/rtw_debug.o
/home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/pci_intf.c: In function ‘pci_dvobj_init’:
/home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/pci_intf.c:514:14: error: implicit declaration of function ‘dma_set_dma_mask’; did you mean ‘dma_set_mask’? [-Werror=implicit-function-declaration]
  514 |         if (!dma_set_dma_mask(pdev, DMA_BIT_MASK(64))) {
      |              ^~~~~~~~~~~~~~~~
      |              dma_set_mask
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/pci_intf.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1842: /home/mauricioquintela/Scripts/rtw8852be] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.4-zen1-1-zen/build'
make: *** [Makefile:607: modules] Error 2
```

lwfinger, 2022-08-27:
I had fixed that error on my system, but had not pushed the fit. Try again.

BTW, when you pulled a lot of stuff, you got all the HRex39 stuff in a directory by that name. It does not work any better than mine.

mauricioquintela, 2022-08-27:
Pulled again, just finished building with no errors. Rebooting now, will edit with status.

Regarding [this](https://github.com/lwfinger/rtw8852be/issues/17#issuecomment-1229167835) comment, what I meant was that installing the HRex39 directory in your repo worked fine, but yours (i.e., main repo) wasnt working. Wording was a bit unclear, that was my fault.

lwfinger, 2022-08-27:
You have a missing package. Install crda and you should be OK. The topic is discussed at https://bbs.archlinux.org/viewtopic.php?id=245962.

closed 2022-08-27 by lwfinger

mauricioquintela, 2022-08-27:
Installed the missing package (weirdly enough, both repositories worked previously without it), modules still not working. 

Ran dmesg again after the two modprobes (modprobe -rv 8852be and modprobe -v 8852be), no errors: [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9438496/dmesg.txt).

mauricioquintela, 2022-08-27:
> You have a missing package. Install crda and you should be OK. The topic is discussed at https://bbs.archlinux.org/viewtopic.php?id=245962.

Installed the missing package (weirdly enough, both repositories worked previously in 5.18.x without it), modules still not working.

Ran dmesg again after the two modprobes (modprobe -rv 8852be and modprobe -v 8852be), no errors: [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9438496/dmesg.txt).

lwfinger, 2022-08-27:
I do not know what caused the problem. I pulled a clean copy, and it failed here too.

I had a bunch of cleanup in my main repo. When I pushed them, and repulled in the test directory, it works now.

mauricioquintela, 2022-08-27:
Was still having issues, so quickly got a random ssd from a drawer to get a clean install going for a sanity check. After going through the dependencies (headers, base-devel, crda), I noticed a new step is present in the README for arch: installing rtw89-dkms-git. After doing that, a clean clone and installation is still failing [dmesg_new.txt](https://github.com/lwfinger/rtw8852be/files/9438789/dmesg_new.txt). I honestly don't know what I am doing wrong...

lwfinger, 2022-08-28:
That dkms package is for the rtw8852ae card. It has no bearing here.

If lspci did not show the b852 device, then it was not available to the PCI bus. That command only depends on the hardware being present. Only the bus drivers built into the kernel are required for that to function.

I was going to install Arch on my system, but after reading the installation instructions, I abandoned that idea. There was too likely a possibility of destroying my main system. That brought me almost back to the days when you downloaded the individual binary blobs onto floppies, and installed what you wanted. I have installed openSUSE, Fedora Ubuntu, Void and Debian that I can remember. All of them have a well-written installation script, which makes me wonder why so many people use Arch. With signed packages, and built-in checksums, there is little danger in installing from binary.

I have no idea why the wifi card does not work on your system. Is it possible that it needs to be reseated. If it is making intermittent connections, that might explain why it did not show up in some lspci output.

mauricioquintela, 2022-08-28:
The dkms is listed in the installation steps for this repo, that's why I mentioned it. I will remove the dkms if it is not for this card.

I agree, the installation process is kinda dumb. Still, have been using it for a few years now and this is the first time I have actual issues.

Regarding wifi card, the part that confuses me the most is that, while installing your main repo is not working, installing from the HRex directory in your repo (with your changes to reduce log spam) works decently.

lwfinger, 2022-08-28:
Try another git pull. It should work.

mauricioquintela, 2022-08-28:
Just did a clean `git clone`, as I noticed there had been tons of changes from the previous version. After rebooting, it is working great! Thank you very much! What exactly was the problem, if I may ask?

lwfinger, 2022-08-28:
I could not find the problem. I copied the HRex39 code, and I will be cleaning it up. According to the driver version, it is a little newer.

mauricioquintela, 2022-08-28:
Just a question: im currently getting my dmesg logs totally spammed by 
```
[  124.801959] Call Trace:
[  124.801959]  <IRQ>
[  124.801959]  pci_unmap_bus_addr+0x36/0x53 [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  core_recycle_txreq_phyaddr+0x6b/0x9f [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  rtw_core_tx_recycle+0x23/0x67 [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  phl_recycle_payload+0x101/0x1d5 [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  _phl_rx_handle_wp_report+0xe4/0xeb [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  phl_rx_pcie+0x2d3/0x45e [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  _phl_rx_callback_pcie+0x9e/0x102 [8852be 4ad89388b2b18e543ee96063cbc8558916a9ca0c]
[  124.801959]  tasklet_action_common.constprop.0+0x132/0x140
[  124.801959]  __do_softirq+0xd3/0x2c9
[  124.801959]  ? handle_edge_irq+0x9a/0x260
[  124.801959]  __irq_exit_rcu+0xbb/0xf0
[  124.801959]  common_interrupt+0x86/0xa0
[  124.801959]  </IRQ>
[  124.801959]  <TASK>
[  124.801959]  asm_common_interrupt+0x26/0x40
```
The rtw_core and phl_rx calls make me thing it is related to this module, but i dont know for sure. Is it related?

lwfinger, 2022-08-29:
That is not one that I have seen. Could I get the full dmesg.txt so that I can see the warning that triggered it?

mauricioquintela, 2022-08-29:
From last boot, as wifi ended up stop working and I had to reboot:
[dmesg_spam.txt](https://github.com/lwfinger/rtw8852be/files/9441733/dmesg_spam.txt)

lwfinger, 2022-08-29:
I decided to abandon the HRex39 version and start from the original code that I was given. It works with no log spamming, and no DMA problems. I get about 325 Mbps download and 350 Mbps upload connecting to a wifi 5 AP in the 5GHz band. I do not have a wifi 6 router for testing.

mauricioquintela, 2022-08-29:
Saved the HRex version in a separate directory, in case I need it for something. I'm now pulling the new version to test it. I believe my router is also wifi 5, so I cant really help with that.

mauricioquintela, 2022-08-29:
Just hit an error while compiling from a clean git clone: 
```
~/Scripts/rtw8852be > make
#rm -f .symvers.8852be
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.4-arch1-1/build M=/home/mauricioquintela/Scripts/rtw8852be  modules
make[1]: Entering directory '/usr/lib/modules/5.19.4-arch1-1/build'
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/platform/platform_linux_pc_pci.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/osdep_service.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/osdep_service_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/rtw_cfg.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/os_intfs.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/xmit_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/mlme_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/recv_linux.o
  CC [M]  /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_cfg80211.o
/home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_cfg80211.c:1202:26: error: ‘struct cfg80211_roam_info’ has no member named ‘links’
 1202 |                 roam_info.links[0].bssid = cur_network->network.MacAddress;
      |                          ^
make[2]: *** [scripts/Makefile.build:249: /home/mauricioquintela/Scripts/rtw8852be/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1847: /home/mauricioquintela/Scripts/rtw8852be] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.4-arch1-1/build'
make: *** [Makefile:643: modules] Error 2
```

mauricioquintela, 2022-08-29:
Oh, I remember that discussion. No changes yet to pull, though

lwfinger, 2022-08-29:
Sorry. I did push it now.

mauricioquintela, 2022-08-29:
Few warning during compilation, but nothing serious. Rebooting now to test. Made sure to `sudo make uninstall` the previous version, just in case. Will edit with results.

mauricioquintela, 2022-08-30:
Module builds and runs perfectly on 5.19.5-Arch, no issues detected. Dmesg logs are still having a "conversation" with 
```
[  154.880596] PHL: [MAC] C2H log: [PST] BcnLossRate = 0/100, BcnTimeoutCnt = 0, BcnErlyCnt = 13
[  154.881130] PHL: [MAC] C2H log: E
[  154.881139] PHL: [MAC] C2H log: <
[  154.883333] PHL: [MAC] C2H log: R
[  154.883564] PHL: [MAC] C2H log: >
[  154.980957] PHL: [MAC] C2H log: LPS Status: 3
```
`BcnErlyCnt` counts to 20 then a bunch of stuff gets dumped: [dmesg_conversations.txt](https://github.com/lwfinger/rtw8852be/files/9452935/dmesg_conversations.txt)