lwfinger/rtw8852be #18: it not work in xubuntu

byronsit, 2022-08-24:
```
xxyy@xxyy:~$ uname -a
Linux xxyy 5.15.0-46-generic #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

xxyy@xxyy:~$ lsmod | grep 88
8852be               4423680  0

nvidia_modeset       1146880  2 nvidia_drm
nls_iso8859_1          16384  1
bluetooth             688128  41 btrtl,btintel,btbcm,bnep,btusb,rfcomm
cfg80211              970752  1 8852be
xxyy@xxyy:~$ sudo modprobe -rv 8852be
[sudo] password for xxyy: 
rmmod 8852be
rmmod cfg80211
xxyy@xxyy:~$ sudo modprobe -v 8852be
insmod /lib/modules/5.15.0-46-generic/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/5.15.0-46-generic/kernel/drivers/net/wireless/realtek/rtw89/8852be.ko 
xxyy@xxyy:~$ 
```

I have tried to reboot many times, and when i use
```
ifconfig -a
```
it shows that

```
enp0s31f6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.102  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::e23a:40e4:1500:277d  prefixlen 64  scopeid 0x20<link>
        ether 98:8f:e0:60:fb:f9  txqueuelen 1000  (Ethernet)
        RX packets 338  bytes 138709 (138.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 476  bytes 137859 (137.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0x5e580000-5e5a0000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 173  bytes 14688 (14.6 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 173  bytes 14688 (14.6 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

i still can not use my wifi.

```
xxyy@xxyy:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 30c9:0057 8SSC21D70376V1SR2395675 Integrated RGB Camera
Bus 003 Device 003: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 005: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 152d:1580 JMicron Technology Corp. / JMicron USA Technology Corp. USB to ATA/ATAPI Bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

```
xxyy@xxyy:~$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:4641] (rev 02)
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:46a6] (rev 0c)
00:04.0 Signal processing controller [1180]: Intel Corporation Device [8086:461d] (rev 02)
00:06.0 PCI bridge [0604]: Intel Corporation Device [8086:464d] (rev 02)
00:06.2 PCI bridge [0604]: Intel Corporation Device [8086:463d] (rev 02)
00:07.0 PCI bridge [0604]: Intel Corporation Device [8086:466e] (rev 02)
00:08.0 System peripheral [0880]: Intel Corporation Device [8086:464f] (rev 02)
00:0a.0 Signal processing controller [1180]: Intel Corporation Device [8086:467d] (rev 01)
00:0d.0 USB controller [0c03]: Intel Corporation Device [8086:461e] (rev 02)
00:0d.2 USB controller [0c03]: Intel Corporation Device [8086:463e] (rev 02)
00:12.0 Serial controller [0700]: Intel Corporation Device [8086:51fc] (rev 01)
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:51ed] (rev 01)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:51ef] (rev 01)
00:15.0 Serial bus controller [0c80]: Intel Corporation Device [8086:51e8] (rev 01)
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:51e0] (rev 01)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:51bc] (rev 01)
00:1c.5 PCI bridge [0604]: Intel Corporation Device [8086:51bd] (rev 01)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:5182] (rev 01)
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Device [8086:51c8] (rev 01)
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:51a3] (rev 01)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:51a4] (rev 01)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (16) I219-V [8086:1a1f] (rev 01)
01:00.0 3D controller [0302]: NVIDIA Corporation Device [10de:25a9] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: Micron Technology Inc Device [1344:5407]
2c:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
2d:00.0 SD Host controller [0805]: O2 Micro, Inc. SD/MMC Card Reader Controller [1217:8621] (rev 01)
xxyy@xxyy:~$ lspci -nn | grep Network
2c:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
xxyy@xxyy:~$ 
```

byronsit, 2022-08-24:
Is any operation  I  still need to do, to enable my wifi?

byronsit, 2022-08-24:
here is my dmesg
```
[    0.853711] pci 0000:00:02.0: reg 0x18: [mem 0x4000000000-0x400fffffff 64bit pref]
[    0.853714] pci 0000:00:02.0: reg 0x20: [io  0x5000-0x503f]
[    0.853728] pci 0000:00:02.0: BAR 2: assigned to efifb
[    0.853728] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.853731] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.853756] pci 0000:00:02.0: reg 0x344: [mem 0x00000000-0x00ffffff 64bit]
[    0.853757] pci 0000:00:02.0: VF(n) BAR0 space: [mem 0x00000000-0x06ffffff 64bit] (contains BAR0 for 7 VFs)
[    0.853760] pci 0000:00:02.0: reg 0x34c: [mem 0x00000000-0x1fffffff 64bit pref]
[    0.853761] pci 0000:00:02.0: VF(n) BAR2 space: [mem 0x00000000-0xdfffffff 64bit pref] (contains BAR2 for 7 VFs)
[    0.855671] pci 0000:00:04.0: [8086:461d] type 00 class 0x118000
[    0.855685] pci 0000:00:04.0: reg 0x10: [mem 0x6103140000-0x610315ffff 64bit]
[    0.857717] pci 0000:00:06.0: [8086:464d] type 01 class 0x060400
[    0.857771] pci 0000:00:06.0: PME# supported from D0 D3hot D3cold
[    0.857791] pci 0000:00:06.0: PTM enabled (root), 4ns granularity
[    0.860027] pci 0000:00:06.2: [8086:463d] type 01 class 0x060400
[    0.860079] pci 0000:00:06.2: PME# supported from D0 D3hot D3cold
[    0.860099] pci 0000:00:06.2: PTM enabled (root), 4ns granularity
[    0.862342] pci 0000:00:07.0: [8086:466e] type 01 class 0x060400
[    0.862407] pci 0000:00:07.0: PME# supported from D0 D3hot D3cold
[    0.862431] pci 0000:00:07.0: PTM enabled (root), 4ns granularity
[    0.862435] pci 0000:00:07.0: DPC: RP PIO log size 0 is invalid
[    0.862438] fbcon: Taking over console
[    0.864869] pci 0000:00:08.0: [8086:464f] type 00 class 0x088000
[    0.864877] pci 0000:00:08.0: reg 0x10: [mem 0x6103195000-0x6103195fff 64bit]
[    0.865862] pci 0000:00:0a.0: [8086:467d] type 00 class 0x118000
[    0.865868] pci 0000:00:0a.0: reg 0x10: [mem 0x6103180000-0x6103187fff 64bit]
[    0.865885] pci 0000:00:0a.0: enabling Extended Tags
[    0.868674] pci 0000:00:0d.0: [8086:461e] type 00 class 0x0c0330
[    0.868683] pci 0000:00:0d.0: reg 0x10: [mem 0x6103170000-0x610317ffff 64bit]
[    0.868722] pci 0000:00:0d.0: PME# supported from D3hot D3cold
[    0.870830] pci 0000:00:0d.2: [8086:463e] type 00 class 0x0c0340
[    0.870839] pci 0000:00:0d.2: reg 0x10: [mem 0x6103100000-0x610313ffff 64bit]
[    0.870846] pci 0000:00:0d.2: reg 0x18: [mem 0x6103194000-0x6103194fff 64bit]
[    0.870875] pci 0000:00:0d.2: supports D1 D2
[    0.870876] pci 0000:00:0d.2: PME# supported from D0 D1 D2 D3hot D3cold
[    0.872824] pci 0000:00:12.0: [8086:51fc] type 00 class 0x070000
[    0.872848] pci 0000:00:12.0: reg 0x10: [mem 0x6103160000-0x610316ffff 64bit]
[    0.872932] pci 0000:00:12.0: PME# supported from D0 D3hot
[    0.875156] pci 0000:00:14.0: [8086:51ed] type 00 class 0x0c0330
[    0.875177] pci 0000:00:14.0: reg 0x10: [mem 0x5e5a0000-0x5e5affff 64bit]
[    0.875262] pci 0000:00:14.0: PME# supported from D3hot D3cold
[    0.876481] pci 0000:00:14.2: [8086:51ef] type 00 class 0x050000
[    0.876504] pci 0000:00:14.2: reg 0x10: [mem 0x610318c000-0x610318ffff 64bit]
[    0.876519] pci 0000:00:14.2: reg 0x18: [mem 0x6103193000-0x6103193fff 64bit]
[    0.879448] pci 0000:00:15.0: [8086:51e8] type 00 class 0x0c8000
[    0.879479] pci 0000:00:15.0: reg 0x10: [mem 0x00000000-0x00000fff 64bit]
[    0.881626] pci 0000:00:16.0: [8086:51e0] type 00 class 0x078000
[    0.881649] pci 0000:00:16.0: reg 0x10: [mem 0x6103191000-0x6103191fff 64bit]
[    0.881736] pci 0000:00:16.0: PME# supported from D3hot
[    0.883965] pci 0000:00:1c.0: [8086:51bc] type 01 class 0x060400
[    0.884062] pci 0000:00:1c.0: PME# supported from D0 D3hot D3cold
[    0.884094] pci 0000:00:1c.0: PTM enabled (root), 4ns granularity
[    0.886358] pci 0000:00:1c.5: [8086:51bd] type 01 class 0x060400
[    0.886455] pci 0000:00:1c.5: PME# supported from D0 D3hot D3cold
[    0.886487] pci 0000:00:1c.5: PTM enabled (root), 4ns granularity
[    0.888747] pci 0000:00:1f.0: [8086:5182] type 00 class 0x060100
[    0.890845] pci 0000:00:1f.3: [8086:51c8] type 00 class 0x040100
[    0.890887] pci 0000:00:1f.3: reg 0x10: [mem 0x6103188000-0x610318bfff 64bit]
[    0.890942] pci 0000:00:1f.3: reg 0x20: [mem 0x6103000000-0x61030fffff 64bit]
[    0.891047] pci 0000:00:1f.3: PME# supported from D3hot D3cold
[    0.893270] pci 0000:00:1f.4: [8086:51a3] type 00 class 0x0c0500
[    0.893294] pci 0000:00:1f.4: reg 0x10: [mem 0x6103190000-0x61031900ff 64bit]
[    0.893325] pci 0000:00:1f.4: reg 0x20: [io  0xefa0-0xefbf]
[    0.894427] pci 0000:00:1f.5: [8086:51a4] type 00 class 0x0c8000
[    0.894446] pci 0000:00:1f.5: reg 0x10: [mem 0xfe010000-0xfe010fff]
[    0.897305] pci 0000:00:1f.6: [8086:1a1f] type 00 class 0x020000
[    0.897347] pci 0000:00:1f.6: reg 0x10: [mem 0x5e580000-0x5e59ffff]
[    0.897569] pci 0000:00:1f.6: PME# supported from D0 D3hot D3cold
[... 682 lines cut ...]
[    9.169842] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
[    9.169845] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
[    9.169848] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
[    9.169851] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
[    9.169853] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
[... cut ...]
[   13.734667] i915 0000:00:02.0: [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[   15.695143] Bluetooth: RFCOMM TTY layer initialized
[   15.695151] Bluetooth: RFCOMM socket layer initialized
[   15.695157] Bluetooth: RFCOMM ver 1.11
[  185.795354] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  185.795458] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'

```

byronsit, 2022-08-24:
I try to use
```
sudo lshw -c network
```
then it shows like this
```
*-network UNCLAIMED
    description: Network Controller
    product: Realtek Semiconductor Co., Ltd.
    vendor: Realtek Semiconductor Co., Ltd.
....
```

I also disable my disable secure boot from BIOS.

lwfinger, 2022-08-24:
Yes, something went wrong with the driver.

I pulled the code from a repo submitted to GitHub from HRex39, and fixed it to build on new kernels. I have been testing it with mixed results. It spams the logs a lot, and when you push it hard such as by running speedtest.net, it runs out of DMA buffers (swiotb entries full), and freezes. I will be looking at that version, which is found in the HRex39 directory of my repo. Do a 'git pull', 'cd HRex39', and the normal make, etc. If dmesg shows the buffer problem, you will need to reboot.

closed 2022-08-24 by lwfinger