lwfinger/rtw8852be #19: make: *** [Makefile:614: install] Error 1

saikatm, 2022-08-27:
cant install on a new system I have no idea what I am missing... 

```
make[1]: Leaving directory '/usr/lib/modules/5.15.63-1-lts/build'
#cp Module.symvers .symvers.8852be
[saikat@linux rtw8852be]$ sudo make install
[sudo] password for saikat: 
install -p -m 644 8852be.ko  /lib/modules/5.15.63-1-lts/kernel/drivers/net/wireless/realtek/rtw89/
install: cannot create regular file '/lib/modules/5.15.63-1-lts/kernel/drivers/net/wireless/realtek/rtw89/': Not a directory
make: *** [Makefile:614: install] Error 1
[saikat@linux rtw8852be]$ 

```

saikatm, 2022-08-27:
seems like the latest commit is causing the issue ??

`install -p -m 644 $(MODULE_NAME).ko  $(MODDESTDIR)realtek/rtw89/
`

lwfinger, 2022-08-27:
It is fixed now. Do another 'git pull'.

saikatm, 2022-08-27:
re downloaded the repo
git pull 
make 
sudo make install

- still the wifi is missing...

lwfinger, 2022-08-27:
Did you reboot, or do a sudo make -v 8852be?

saikatm, 2022-08-27:
ofcourse I did

> Did you reboot, or do a sudo modprobe -v 8852be?

saikatm, 2022-08-28:
pull the latest [commit](https://github.com/lwfinger/rtw8852be/commit/6be9eb2e93708287bcd7d830bcb885e62acb0efe) it doesnt work too.

saikatm, 2022-08-28:
also tried [this](https://github.com/lwfinger/rtw8852be/commit/77b7e10e215f58a334a59832c58feb3577d4e402) not working!!!

saikatm, 2022-08-28:
@lwfinger the machine is unusable please help

lwfinger, 2022-08-28:
I found a problem. Do another 'git pull' and remake.

saikatm, 2022-08-28:
@lwfinger just did but here are the issue.
- it does not work anymore in the LTS kernel  
- worked for few minutes on the current kernel.
- stopped working after few minutes, rebooted wifi gone
- again did sudo make install then sudo modprove 8852be give this error
```
 [saikat@linux ~]$ sudo modprobe 8852be 
[sudo] password for saikat: 
modprobe: ERROR: could not insert '8852be': Exec format error
```

saikatm, 2022-08-28:
@lwfinger just did but here are the issue.
- it does not work anymore in the LTS kernel  
- worked for few minutes on the 5.19.4 kernel.
- stopped working after few minutes, rebooted wifi gone
- again did sudo make install wifi is missing

```

lwfinger, 2022-08-28:
Your complaints provide me with little information. For example, what version is the LTS kernel, and are there any error messages. When the driver stops working, it needs a cold reboot with the power off for a few seconds.

The driver is known to have problems with running out of DMA buffers. Running speedtest.net is certain to crash it, and there is a report that it fails under normal loads after a long time. No, I do not know how to fix it. To see if that is the problem, then look at the output of dmesg. DO NOT ATTACH IT HERE. I have seen lots of those errors on my system.

You have several options. As this is likely a new laptop, you could return it. A second option would be to buy an inexpensive USB wifi chip such as the atheros ar9271. A third option would require opening the laptop, finding the format of the wifi chip, and finding another vendor's product with that format.

closed 2022-08-28 by lwfinger

saikatm, 2022-08-29:
Thanks for your brief replay. 
the driver worked fine until few recent commits.
I think I have to use windows for some times until it gets fixed.
do you think realtek will release a driver for the chip in future?

lwfinger, 2022-08-29:
Well, this driver came from Realtek What do you mean?

If it works until recent commits, check out the earlier version. That is the beauty of git. You can also bisect through the changes.

lwfinger, 2022-08-29:
Do another pull.

saikatm, 2022-08-29:
@lwfinger DID NOT WORK ON THE LTS KERNEL.
getting this error while compiling on the curren 
```
saikat@linux ~/r/rtw8852be (main)> make
#rm -f .symvers.8852be
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.4-arch1-1/build M=/home/saikat/repos/rtw8852be  modules
make[1]: Entering directory '/usr/lib/modules/5.19.4-arch1-1/build'
  CC [M]  /home/saikat/repos/rtw8852be/platform/platform_linux_pc_pci.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/osdep_service.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/osdep_service_linux.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/rtw_cfg.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/os_intfs.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/ioctl_linux.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/xmit_linux.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/mlme_linux.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/recv_linux.o
  CC [M]  /home/saikat/repos/rtw8852be/os_dep/linux/ioctl_cfg80211.o
/home/saikat/repos/rtw8852be/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/saikat/repos/rtw8852be/os_dep/linux/ioctl_cfg80211.c:1202:26: error: ‘struct cfg80211_roam_info’ has no member named ‘links’
 1202 |                 roam_info.links[0].bssid = cur_network->network.MacAddress;
      |                          ^
make[2]: *** [scripts/Makefile.build:249: /home/saikat/repos/rtw8852be/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1847: /home/saikat/repos/rtw8852be] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.4-arch1-1/build'
make: *** [Makefile:643: modules] Error 2
saikat@linux ~/r/rtw8852be (main) [2]> 
```

lwfinger, 2022-08-29:
That is now fixed. Do another pull.

saikatm, 2022-08-29:
@lwfinger thanks working fine now. did an speedtest.net did not had any issue. will report if any.