lwfinger/rtw8852be #20: Does not work on Linux 6 Manjaro(and ubuntu, popos)

ZykaJiri, 2022-08-28:
Hello, the driver does not seem to work on Linux 6 it installs and gets inicialised without any problems but the wifi status does not change. Tried on multiple distributions - Ubuntu, PopOS, Manjaro

Always did a full reboot between reinstalls/configurations. Tried specifying new_id to the wifi module id on Ubuntu, this did not help.
Bluetooth works fine

## Debugging logs
Neofetch:
[image]

dmesg: [dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9439741/dmesg.txt)

lshw: [lshw.txt](https://github.com/lwfinger/rtw8852be/files/9439756/lshw.txt)

lspci -nn: [lspci.txt](https://github.com/lwfinger/rtw8852be/files/9439757/lspci.txt)

## Command outputs
sudo modprobe -v 8852be
> Outputs nothing

sudo modprobe -vr 8852be && sudo modprobe -v 8852be
> rmmod 8852be
> rmmod cfg80211
> insmod /lib/modules/6.0.0-1-MANJARO/kernel/net/wireless/cfg80211.ko.zst 
> insmod /lib/modules/6.0.0-1-MANJARO/kernel/drivers/net/wireless/realtek/rtw89/8852be.ko 

I hope i provided all the possible relevant information. May have missed something. Thank you for reading this far.

lwfinger, 2022-08-28:
I had a problem. Do another 'git pull', and all should be OK.

ZykaJiri, 2022-08-28:
That did it. Thank you very much!

closed 2022-08-28 by ZykaJiri

lwfinger, 2022-08-29:
Do another pull. I fixed the DMA issue.