lwfinger/rtw8852be #23: Showing adapter but can't find wifi networks (Kernel 5.19.0-76051900)

codecent, 2022-09-05:
Hey, I'm pretty new to Linux. 

I'm on recent PopOS, Kernel version 5.19.0-76051900.

I installed everything as i'm supposed to, but getting the "skipping btf generation due to unavailabililty of vmlinux.xz" during "make".

I installed it anyway with following output: 

```
install -p -m 644 8852be.ko  /lib/modules/5.19.0-76051900-generic/kernel/drivers/net/wireless/realtek/rtw89/
/sbin/depmod -a 5.19.0-76051900-generic
```

Wifi card is now shown, but can't find networks straight away. Sometimes after disabling and reenabling it works, but isn't stable. 

Also installed bluetooth drivers, same "Skipping BTF generation"-error during "make", but then seems to install correctly afterwards. Doesnt show any devices tho. 

unloading and loading wifi driver is giving: 
```
rmmod 8852be
rmmod cfg80211
```
```
insmod /lib/modules/5.19.0-76051900-generic/kernel/net/wireless/cfg80211.ko 
insmod /lib/modules/5.19.0-76051900-generic/kernel/drivers/net/wireless/realtek/rtw89/8852be.ko 

```
[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9492154/dmesg.txt)

lspci -knn shows:
```

03:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b852]
	Subsystem: Lenovo Device [17aa:4853]
	Kernel driver in use: rtl8852be
	Kernel modules: 8852be

```

Is there any way to fix it?

lwfinger, 2022-09-06:
Ignore the BTF warning. That is harmless. Your dmesg.txt shows a warning
[  777.601191] UBSAN: array-index-out-of-bounds in /home/vincent/rtw8852be/phl/phl_cmd_dispatcher.c:563:5
[  777.601194] index 16 is out of range for type 'u8 [16]'
I have pushed a fix that should work for that. Do a 'git pull', 'make', and 'sudo make install', and then reboot.

Another error had just rolled off the top of your dmesg buffer. Please run the dmesg > dmesg.txt a lot sooner. Your system had been up for 777 seconds when the oldest entry had been generated, and 801 sec when you made the dump. Do the dmesg dump shortly after the system boot.

codecent, 2022-09-06:
Bluetooth seems to work now. Feels slow in finding devices and connecting to them, but it works. 

Wifi still doesn't show any networks. I created dmesg.txt shortly after reboot now:

lwfinger, 2022-09-06:
Your dmesg buffer is quite small and it rolled over by 22 seconds after reboot.

I reduced the amount of log spamming so that real errors will stand out.

Your kernel has KASAN memory debugging enabled, which checks for array overruns. There were several places that needed fixing. The funny thing about KASAN errors is that if the kernel is generated with it off, the problems are ignored; hoever, with it on, the overrun is not ignored and the device fails.

I had to enable KASAN in my kernel to find the problems. Do a 'git pull' and you should be good.

lwfinger, 2022-09-06:
By the way, with KASAN enabled, the speedtest results to a local server were between 60 and 100 Mbps. With KASAN disabled, that jumped up to 350 Mbps. If you are generating your own kernel, CONFIG_KASAN should be set to "n" in your kernel configuration. If you are using a standard kernel, see if your distro has a kernel with it turned off. Due to the significant performance hit, most distros have this turned off.

codecent, 2022-09-06:
I did 'git pull' etc., but it didn't fix it. 
I'm using a standard PopOS Kernel right now. I will see if I can figure out changing to another version in future, thank you for the information and anway for all your help. 

[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9498701/dmesg.txt)

I think I took it very soon after startup, should I wait a little longer?

lwfinger, 2022-09-06:
No, that is fine. The errors from the wifi device are now gone. You have some warnings from other devices, but the only output for the 8852be is the renaming of the wlan0 device to wlp3s0, which is the name you need to use to make a connection.

I am using NetworkManager with the KDE applet, and the connection works straight away. There seems to be some problem with your network connection stuff. If you do 'sudo iw dev wlp3s0 scan', do you see the neighboring APs?

codecent, 2022-09-06:
It's different. First time I tried, it said "device is busy", second attempt didn't do anything. Third attempt actually showed APs and let me connect.
Then i restarted to try again. It started with "device is busy again" and all further attempt theres no output at all. 

I also recognized major audio lags from time to time, eventually everytime its scanning for APs in the background. When wifi is disabled, audio lag is gone. When i reenable it, it's back.

Here is my dmesg after trying: 

[dmesg.txt](https://github.com/lwfinger/rtw8852be/files/9499005/dmesg.txt)

lwfinger, 2022-09-06:
The "device is busy" comes about because cfg80211 is in the middle of doing a scan.

When I was running the kernel with KASAN enabled, I not only had slow rates, but the connection dropped with nothing logged.

You need to get a new kernel with KASAN disabled by getting it from PopOS, generating your own, or switching to another distro. I looked at the PopOS web page to see what kind of desktop they have. To me, it looks similar to Xfce or some other similar desktop. If you were to install openSUSE Leap 15.4, that is one of the options.

In any case, as long as you are using a KASAN-enabled kernel, there is nothing I can do to help you.

closed 2022-09-06 by lwfinger

codecent, 2022-09-06:
Ok, thank you for your work, I'll change to openSuse!