lwfinger/rtw8852be #25: UBSAN warnings spammed for shift-out-of-bounds on Yoga 6 (13",7)

whitty, 2022-09-17:
Many thanks, I've just gotten WiFi up on my new laptop - Ubuntu 22.04 - seems to be working OK, but on module load I get spammed by some ugly looking warnings.  Not sure if this is normal, or peculiar to my hardware/setup so thought I'd raise them.

I haven't looked into the code yet to see if its anything obvious.

h2. In open

```
[    4.242884] ================================================================================
[    4.242892] UBSAN: shift-out-of-bounds in /var/lib/dkms/rtw8852be/1.0.0/build/phl/hal_g6/phy/bb/halbb_interface.c:235:42
[    4.242899] shift exponent 32 is too large for 32-bit type 'unsigned int'
[    4.242905] CPU: 7 PID: 661 Comm: NetworkManager Tainted: G           O      5.15.0-47-generic #51-Ubuntu
[    4.242908] Hardware name: LENOVO 82UD/LNVNB161216, BIOS JECN11WW(V1.04) 03/14/2022
[    4.242910] Call Trace:
[    4.242912]  <TASK>
[    4.242915]  show_stack+0x52/0x5c
[    4.242923]  dump_stack_lvl+0x4a/0x63
[    4.242928]  dump_stack+0x10/0x16
[    4.242931]  ubsan_epilogue+0x9/0x49
[    4.242934]  __ubsan_handle_shift_out_of_bounds.cold+0x61/0xef
[    4.242940]  ? halbb_set_cr+0x41/0x7d [8852be]
[    4.243041]  ? pci_read32+0x15/0x1b [8852be]
[    4.243162]  ? _hal_read32+0x34/0x3d [8852be]
[    4.243281]  halbb_get_reg_cmn+0x5d/0x5f [8852be]
[    4.243376]  halbb_dig_gain_para_init+0x28/0x655 [8852be]
[    4.243462]  halbb_dig_init+0x8c/0xff [8852be]
[    4.243541]  halbb_dm_init+0x9c/0x17d [8852be]
[    4.243629]  rtw_hal_bb_dm_init+0x1a/0x20 [8852be]
[    4.243749]  hal_start_8852b+0x134/0x24a [8852be]
[    4.243853]  hal_start_8852be+0x58/0xa0 [8852be]
[    4.243950]  ? __nla_validate_parse+0x152/0x1a0
[    4.243957]  rtw_hal_start+0x33/0x10d [8852be]
[    4.244078]  rtw_phl_start+0x35/0x14f [8852be]
[    4.244183]  rtw_hw_start+0x26/0x61 [8852be]
[    4.244280]  netdev_open+0xa3/0x188 [8852be]
[    4.244353]  __dev_open+0xf7/0x1c0
[    4.244359]  __dev_change_flags+0x1a7/0x220
...
[    4.244509] RIP: 0033:0x7f76f7e14b4d
[    4.244513] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 3a 8f f6 ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 33 44 89 c7 48 89 44 24 08 e8 7e 8f f6 ff 48
[    4.244515] RSP: 002b:00007fffdf9f7eb0 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
[    4.244519] RAX: ffffffffffffffda RBX: 000000000000000c RCX: 00007f76f7e14b4d
[    4.244521] RDX: 0000000000000000 RSI: 00007fffdf9f7ef0 RDI: 000000000000000c
[    4.244523] RBP: 0000558932968040 R08: 0000000000000000 R09: 0000000000000000
[    4.244524] R10: 0000000000000000 R11: 0000000000000293 R12: 0000000000000000
[    4.244526] R13: 00007fffdf9f8040 R14: 00007fffdf9f803c R15: 0000000000000000
```

```
[    4.244590] ================================================================================
[    4.244593] UBSAN: shift-out-of-bounds in /var/lib/dkms/rtw8852be/1.0.0/build/phl/hal_g6/phy/bb/halbb_interface.c:184:44
[    4.244598] shift exponent 32 is too large for 32-bit type 'unsigned int'
[    4.244602] CPU: 7 PID: 661 Comm: NetworkManager Tainted: G           O      5.15.0-47-generic #51-Ubuntu
[    4.244605] Hardware name: LENOVO 82UD/LNVNB161216, BIOS JECN11WW(V1.04) 03/14/2022
[    4.244606] Call Trace:
[    4.244607]  <TASK>
[    4.244608]  show_stack+0x52/0x5c
[    4.244612]  dump_stack_lvl+0x4a/0x63
[    4.244615]  dump_stack+0x10/0x16
[    4.244618]  ubsan_epilogue+0x9/0x49
[    4.244621]  __ubsan_handle_shift_out_of_bounds.cold+0x61/0xef
[    4.244625]  ? halbb_set_cr+0x41/0x7d [8852be]
[    4.244720]  ? halbb_set_reg_cmn+0x30/0x92 [8852be]
[    4.244806]  halbb_set_reg_cmn+0x90/0x92 [8852be]
[    4.244889]  halbb_sdagc_follow_pagc_config+0x64/0x111 [8852be]
[    4.244975]  halbb_dig_cfg_bbcr+0x17f/0x184 [8852be]
[    4.245055]  halbb_dig_init+0xc8/0xff [8852be]
[    4.245134]  halbb_dm_init+0x9c/0x17d [8852be]
[    4.245220]  rtw_hal_bb_dm_init+0x1a/0x20 [8852be]
[    4.245269]  hal_start_8852b+0x134/0x24a [8852be]
[    4.245269]  hal_start_8852be+0x58/0xa0 [8852be]
[... 110 lines cut ...]
[    4.444245]  ubsan_epilogue+0x9/0x49
[    4.444247]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[    4.444249]  ? atkbd_interrupt+0x471/0x840
[    4.444252]  ? atkbd_interrupt+0x471/0x840
[    4.444254]  get_module_by_id+0x12c/0x160 [8852be]
[    4.444343]  dispr_send_msg+0x28a/0x34b [8852be]
[    4.444430]  phl_disp_eng_send_msg+0x73/0x7a [8852be]
[    4.444517]  _phl_cmd_scan_req_acquired+0x13a/0x184 [8852be]
[    4.444609]  register_cur_cmd_req+0x40/0x6a [8852be]
[    4.444697]  dispr_process_token_req+0x5a/0xe1 [8852be]
[    4.444783]  _handle_token_op_info+0x82/0x16f [8852be]
[    4.444869]  token_op_hanler+0x33/0x93 [8852be]
[    4.444954]  dispr_thread_loop_hdl+0x3d/0x1a8 [8852be]
[    4.445039]  dispr_share_thread_loop_hdl+0xe/0x14 [8852be]
[    4.445124]  share_thread_hdl+0x6a/0xfe [8852be]
[    4.445213]  ? phl_ps_hal_pwr_req+0x8f/0x8f [8852be]
[    4.445307]  kthread+0x12a/0x150
[    4.445309]  ? set_kthread_struct+0x50/0x50
[    4.445316]  ret_from_fork+0x22/0x30
[    4.445316]  </TASK>
[    4.445338] ================================================================================

```

Apologies if this is just known, benign issues.

whitty, 2022-09-17:
hmm,... - the dispatcher errors are `array-index-out-of-bounds: index 16 is out of range for type 'u8 [16]'` in a number of places the `bitmap` is defined against `PHL_BK_MDL_END` (128), but the incoming values for `mdl_id` is 131.  There's a `PHL_MDL_ID_MAX` which is 255 - not sure if that means the wrong ids are being handled in the wrong place, or if the size is just wrong?

What is `BK` in this case?

lwfinger, 2022-09-17:
Those KASAN warnings are not critical - just alarming.

I just pushed a fix for the ones you listed.

whitty, 2022-09-18:
Thanks for the attention - I think the bitmap out-of-range issues are potentially an issue (its really checking the next struct member)

```
if (mdl_id % 8 >= 16)
```

That condition is mathematically impossible - right?

The check I'd use would be

```
if (mdl_id >= PHL_BK_MDL_END)
```

of if you wanted it oriented on the container

```
if (mdl_id / 8 >= sizeof(obj->bitmap))
```

whitty, 2022-09-18:
Not wanting to kibbitz (I do a bunch of code-review for a living ;-) )

```
	if (shift < 32)		/* max shift allowed is 31 */
		val_mod = ((ori_val) & (~mask)) | (((val << shift)) & mask);
```

You can get a minor improvement (although a good optimiser probably figures it out) by realising the situation where `shift == 32` is strictly the case where `mask == 0`, thus all the `& mask` expressions just go to zero - ie in most cases you can just do this on the first line:

```
   if (mask == 0)
       return 0;
```

I was a little concerned as to why mask would ever validly be passed 0 though (has code smell about it)

whitty, 2022-09-19:
```
	if (mask < 32)		/* max shift allowed is 31 */
		val_0 = (halbb_get_cr(bb, addr) & mask) >> halbb_cal_bit_shift(mask);
```

Also that should be `(halbb_cal_bit_shift(mask) < 32)` - that does cause a true failure

Although I think 

```
	if (mask != 0)
		val_0 = (halbb_get_cr(bb, addr) & mask) >> halbb_cal_bit_shift(mask);
```

 probably works just as well

lwfinger, 2022-09-19:
My usual kernel does not have KASAN checks enabled. I have reverted my patch, and look forward to seeing yours.

whitty, 2022-09-19:
Cheers, I'm working through the pieces - there's something I don't understand WRT the ID's in the dispatcher - I'm worried it worked by "coincidence" before, ie bit 131 overlapped bit 5 in the next bitmap.

Because following your lead of considering out of range bitmap entries as zero does seem to be confusing things (even though it seems the most sane thing to do).

When I had it working I had simply expanded the bitmaps to cover the range of the ID 131.

Sorry for the bother this caused other people :-(

lwfinger, 2022-09-19:
I think expanding the range is a better way to go. Obviously, returning 0 breaks things.

whitty, 2022-09-20:
Sorry - been moving a bit slower recently - I was seeing `WARN()`s from the IEEE stack as I tried adding/removing different approaches, but eventually figured out its related to dynamically loading and unloading the module without rebooting.  Since then I've been running for long periods with no indication of error/warning when loading the module at boot time, and never unloading it.

Now I've been filtering changes down to smallest possible set (because its not clear why things are quite as they are - and if its not broke I don't want to touch it,...).  PRs soon.

whitty, 2022-09-28:
OK - the root causes are

`halbb_dig_gain_para_init()`  accesses a bunch of (~20) registers that are `#ifdef`'d out in `halbb_cr_cfg_dig_init()` selected as `cr_type == 3 (BB_CLIENT)`

`halbb_sdagc_follow_pagc_config()` accesses 4 registers that are likewise ifdef'd out  in `halbb_cr_cfg_dig_init()`

Any idea why those registers are ifdef'd out?  They're ifdef'd out in the initial import.

whitty, 2022-09-28:
ok - they're ifdef'd out because the defs don't exist :-/

lwfinger, 2022-09-28:
I do not have a clue, other than they may be needed in some other device that shares most of this driver.

Some good news. Realtek has started pushing rtw8852b code to the kernel. They hope to have the driver in kernel 6.2. As the code gets merged into the wireless-next kernel, I will be applying it to my rtw89 repo and testing there.

closed 2022-12-09 by lwfinger