lwfinger/rtw8852be #26: Wifi keep disconnecting

Bananeqqq, 2022-09-18:
Hi,
idk if it's issue on my side, but the wifi keeps disconnecting and until I turn it off and on, I don't even see any networks available. What should I do?

Btw, thanks for this. I was looking for that driver for an hours and finally...

saikatm, 2022-09-18:
yes same issue here. bith lts and normal kernel

lwfinger, 2022-09-19:
What do the logs say about the disconnects?

Bananeqqq, 2022-09-19:
Idk if it's what you want, but I tried journalctl -u NetworkManager.service (I am new to Ubuntu and all this, sry xd), and https://pastebin.com/UieLuTwc

Warns
<warn>  [1663513298.8872] device (wlp1s0): Activation: (wifi) association took too long, failing activation
<warn>  [1663513298.8883] device (wlp1s0): Activation: failed for connection
<warn>  [1663513589.8864] device (wlp1s0): link timed out.
<warn>  [1663513640.8882] device (wlp1s0): Activation: (wifi) association took too long, failing activation

And those repeated.

But in the evening it went like for an hours without disconnecting as well as right now in the morning it's fine, so idk what I did, but rn it seems working.

Bananeqqq, 2022-09-19:
After an almost hour it again disconnected 

zář 19 08:36:00 bananeq NetworkManager[685]: <info>  [1663569360.3539] device (wlp1s0): supplicant interface state: completed -> disconnected
zář 19 08:36:00 bananeq NetworkManager[685]: <info>  [1663569360.3540] device (p2p-dev-wlp1s0): supplicant management interface state: completed -> disconnected
zář 19 08:36:00 bananeq NetworkManager[685]: <info>  [1663569360.4540] device (wlp1s0): supplicant interface state: disconnected -> scanning
zář 19 08:36:00 bananeq NetworkManager[685]: <info>  [1663569360.4541] device (p2p-dev-wlp1s0): supplicant management interface state: disconnected -> scanning
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.0100] device (wlp1s0): supplicant interface state: scanning -> associating
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.0101] device (p2p-dev-wlp1s0): supplicant management interface state: scanning -> associating
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3325] device (wlp1s0): supplicant interface state: associating -> 4way_handshake
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3326] device (p2p-dev-wlp1s0): supplicant management interface state: associating -> 4way_handshake
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3326] device (wlp1s0): ip:dhcp4: restarting
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3682] dhcp4 (wlp1s0): canceled DHCP transaction
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3683] dhcp4 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3684] dhcp4 (wlp1s0): state changed no lease
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3686] dhcp4 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3694] device (wlp1s0): supplicant interface state: 4way_handshake -> completed
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.3695] device (p2p-dev-wlp1s0): supplicant management interface state: 4way_handshake -> completed
zář 19 08:36:01 bananeq NetworkManager[685]: <info>  [1663569361.4041] dhcp4 (wlp1s0): state changed new lease, address=10.50.109.206
zář 19 08:36:26 bananeq NetworkManager[685]: <info>  [1663569386.2055] policy: set 'Sincoolka 5G' (wlp1s0) as default for IPv6 routing and DNS
zář 19 08:36:26 bananeq NetworkManager[685]: <warn>  [1663569386.2084] dns-sd-resolved[f2c58ab5a7ae10a8]: send-updates SetLinkDomains@3 failed: GDBus.Error:org.freedesktop.resolve1.NoSuchLink: Link 3 not known
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9053] device (wlp1s0): supplicant interface state: completed -> associating
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9054] device (p2p-dev-wlp1s0): supplicant management interface state: completed -> associating
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9063] device (wlp1s0): ip:dhcp4: restarting
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9439] dhcp4 (wlp1s0): canceled DHCP transaction
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9440] dhcp4 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9440] dhcp4 (wlp1s0): state changed no lease
zář 19 08:37:37 bananeq NetworkManager[685]: <info>  [1663569457.9441] dhcp4 (wlp1s0): activation: beginning transaction (timeout in 45 seconds)
zář 19 08:37:45 bananeq NetworkManager[685]: <info>  [1663569465.9530] device (wlp1s0): supplicant interface state: associating -> disconnected
zář 19 08:37:45 bananeq NetworkManager[685]: <info>  [1663569465.9531] device (p2p-dev-wlp1s0): supplicant management interface state: associating -> disconnected
zář 19 08:37:46 bananeq NetworkManager[685]: <info>  [1663569466.0585] device (wlp1s0): supplicant interface state: disconnected -> scanning
zář 19 08:37:46 bananeq NetworkManager[685]: <info>  [1663569466.0586] device (p2p-dev-wlp1s0): supplicant management interface state: disconnected -> scanning
zář 19 08:37:50 bananeq NetworkManager[685]: <info>  [1663569470.3737] device (wlp1s0): supplicant interface state: scanning -> associating
zář 19 08:37:50 bananeq NetworkManager[685]: <info>  [1663569470.3738] device (p2p-dev-wlp1s0): supplicant management interface state: scanning -> associating
zář 19 08:37:50 bananeq NetworkManager[685]: <info>  [1663569470.8285] device (wlp1s0): supplicant interface state: associating -> disconnected
zář 19 08:37:50 bananeq NetworkManager[685]: <info>  [1663569470.8286] device (p2p-dev-wlp1s0): supplicant management interface state: associating -> disconnected
zář 19 08:37:51 bananeq NetworkManager[685]: <info>  [1663569471.3291] device (wlp1s0): supplicant interface state: disconnected -> scanning
zář 19 08:37:51 bananeq NetworkManager[685]: <info>  [1663569471.3291] device (p2p-dev-wlp1s0): supplicant management interface state: disconnected -> scanning
zář 19 08:37:51 bananeq NetworkManager[685]: <info>  [1663569471.9778] device (wlp1s0): supplicant interface state: scanning -> associating
zář 19 08:37:51 bananeq NetworkManager[685]: <info>  [1663569471.9779] device (p2p-dev-wlp1s0): supplicant management interface state: scanning -> associating
[... cut ...]
zář 19 08:38:01 bananeq NetworkManager[685]: <info>  [1663569481.3260] device (p2p-dev-wlp1s0): supplicant management interface state: disconnected -> scanning
zář 19 08:38:01 bananeq NetworkManager[685]: <warn>  [1663569481.5333] device (wlp1s0): link timed out.
zář 19 08:38:01 bananeq NetworkManager[685]: <info>  [1663569481.5335] device (wlp1s0): state change: activated -> failed (reason 'supplicant-timeout', sys-iface-state: 'managed')
zář 19 08:38:01 bananeq NetworkManager[685]: <info>  [1663569481.5340] manager: NetworkManager state is now DISCONNECTED

stingbo, 2022-11-12:
same issue here

lwfinger, 2022-11-12:
Do not use this repo. This driver is much inferior to the one at https://github.com/lwfinger/rtw89.git.

As this one tells you every time you build it, this repo will be deleted on Dec. 1. I am no longer maintaining it, or trying to find fixes.

closed 2022-11-12 by lwfinger