lwfinger/rtw8852be #32: WiFi unstable and drops intermittently

BenOoi, 2022-10-08:
Assuming my new HP 835 G9 would come with the RTL8852AE hoping for good linux support, to find out it was the BE model. So I had to install this kernel module.

This is the one thing that is unstable on my laptop and hope I can get your help to make it more stable. Reproducable problems:

- ping seems high, 9-10ms to my first next hop (gateway of my laptop) and sometimes 1second+ response time spikes
Sitting with my HP chromebook on the same spot gives me 2-3ms. 200ms spikes
- Dropping connection with dmesg saying: 
rtw8852be: Connection disconnected for reason 23
rtw8852be: Connection disconnected for reason 0
rtw8852be: Connection disconnected for reason 7
Numbers seem to differ from time to time, but 0 and 1 are the most common.

I also had dmesg filling up when I ran iperf, but cannot reproduce that anymore. Also iperf speed seems slow at 150mbit/s, when my phone easily reaches 340mbit/s.

System details:

 ~ sudo lspci -vv -s 01:00 
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
	Subsystem: Hewlett-Packard Company Device 88e3
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 32 bytes
	Interrupt: pin A routed to IRQ 121
	IOMMU group: 11
	Region 0: I/O ports at 6000 [size=256]
	Region 2: Memory at b4a00000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fee00000  Data: 0000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 75.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis+ LTR+ OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Device Serial Number 00-e0-4c-ff-fe-***
	Capabilities: [158 v1] Latency Tolerance Reporting
		Max snoop latency: 1048576ns
		Max no snoop latency: 1048576ns
	Capabilities: [160 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=30us PortTPowerOnTime=60us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: rtl8852be
	Kernel modules: 8852be

 ~  uname -a 
Linux hhwlnx004 5.19.13-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Oct 4 15:42:43 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Thanks for your support!

lwfinger, 2022-10-08:
Unfortunately, what you see is the best we can give at the moment. The rtw8852be driver as part of rtw89 is nearing completion. I have an early release that I am not allowed to publish. It is stable and yields about 500 Mbps when communicating with a local speedtest instance through a Wifi5 AP. Speeds would be even higher if I had a Wifi6 AP.

Looking at site https://aboutcher.co.uk/2012/07/linux-wifi-deauthenticated-reason-codes/
Reason 0 is reserved
Reason 1 = Unspecific Reason | We don’t know what’s wrong
Reason 7 = Class 3 frame received from nonassociated STA | Client attempted to transfer data before it was associated.
Reason 23 = IEEE 802.1X authentication failed

What is the make/model of your AP?
Please run 'iw dev', followed by 'sudo iv dev <YourDevice> scan > scan.txt' where you substitute the name of your device from the first command into the <YourDevice> field. Attach scan.txt to this issue.

mauricioquintela, 2022-10-08:
I had some issues with sudden ping spikes to above 5000ms a few weeks ago which were making video calls unusable. I just assumed they were caused by crappy internet (eduroam is always setup stupidly), so I didn't pay much attention to them. Where I am currently only has wired internet, so I have not tested it any further.  

Glad to hear that rtw89 support for this card is nearing completion!

BenOoi, 2022-10-10:
Good to hear rtw89 will fix this. I honestly don't care about the speed, stability is more important to me. Some info from me:

home: Unifi AP AC Lite and LR, wifi 5 AC wave 1
work: Extreme AP305C, wifi 6

Scan from home network:
https://gist.github.com/BenOoi/dd0278e1644a62b276c92eefa6ca6d7c

lwfinger, 2022-10-10:
The remaining pieces of the official driver for 8852be were pushed to wireless-next yesterday. I added them to my copy of the repo and built the driver, which has been running on my computer for about 19 hours. During that time, it has not dropped the connection. I plan to push it in about 3 hours. The next time there is an update for rtw89, it will be there. By about Dec. 1, I hope to remove all other sources for 8852be drivers.

BenOoi, 2022-11-03:
I upgraded to the rtw89 driver. Seems to be more stable. Iperf is not creating messages in dmesg anymore. Will test this more the following days.

lwfinger, 2022-11-03:
The driver in rtw89 is definitely superior.

closed 2022-11-03 by lwfinger