lwfinger/rtw8852be #33: Ubuntu 22.04 and kernel 6.0: make error

BruneWayce, 2022-10-09:
Hello,
as in the subject, in a fresh new install I get first this error:
`gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’`

I've found somewhere that in order to solve I have to  add this:
`export LLVM=1 && export LLVM_IS=1`

Then I get this:
```
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           
/bin/sh: 1: clang: not found
```
So I installed clang, but now:
```
clang: error: unknown argument: '-mfunction-return=thunk-extern'
clang: error: unknown argument: '-fzero-call-used-regs=used-gpr'
clang: error: unknown argument: '-fconserve-stack'
clang: error: unsupported option '-mrecord-mcount' for target 'x86_64-unknown-linux-gnu'
clang: error: '-ftrivial-auto-var-init=zero' hasn't been enabled; enable it at your own peril for benchmarking purpose only with '-enable-trivial-auto-var-init-zero-knowing-it-will-be-removed-from-clang'
```
At this point I think that I'm getting too many errors, for sure there is something wrong in my way of solving the problem of the missing driver for RTL8852be..
Any help?

lwfinger, 2022-10-10:
You need to use gcc, not clang. If you have the gcc used to build the kernel (12.2.0), it will know what to do with those options. By the way, those options come from the kernel, not this driver.

BruneWayce, 2022-10-10:
Thank you so much! With gcc 12.2.0 everything went ok, also "export LLVM=1 && export LLVM_IS=1" wasn't necessary.

Problem solved and wifi working.
I was already ready with an AX210 in the Amazon cart... ;-)

closed 2022-10-10 by BruneWayce