lwfinger/rtw8852be #36: This repo works for me but not rtw89

michaelphamct, 2022-10-27:
Hello,

I see in the latest commit message that this repo is being deprecated in favor of the one at https://github.com/lwfinger/rtw89.git . However, lwfinger/rtw89 did not fix my missing Wi-Fi issue on my laptop but lwfinger/rtw8852be did give me working Wi-Fi. Please keep this repo open until at least the end of December when I will have more free time to test lwfinger/rtw89 again. At the moment, I can't afford to be debugging my Wi-Fi issues until my school semester is over.

lwfinger, 2022-10-28:
When I delete the repo, your copy will still exist; however, you will get no support, which will not be that big a loss. I am not supporting it anyway.

If you tried the rtw89 repo more than a week ago, then you did not get the current rtw8852be driver. The new one is a lot smaller and faster than the one you are now using. Connected to an ax1500 (wifi6) AP, I get 760 Mbps down and 690 Mbps up.

michaelphamct, 2022-10-28:
I just tried lwfinger/rtw89 again and this time it worked. However, my performance dropped compared to using lwfinger/rtw8852be. Previously, rtw8852be would give me around around ~700 Mbps down but rtw89 is giving me around ~500 Mbps down. The latency also seems to be affected. I am getting ~18 ms latency on rtw89 but ~8ms on rtw8852be. So overall, rtw89 seems to be slower for me than rtw8852be. Although the tests don't seem to be too consistent anyway so I am not too concerned as long as it is working. Feel free to close this issue. Thanks!

michaelphamct, 2022-10-28:
This is an unrelated comment but do you know when rtw89 will be upstreamed in mainline kernel? While the rtw89 git repo works, I had to use an Ethernet cable to get to your GitHub and of course having Wi-Fi working out of the box will be a much better experience. :)

lwfinger, 2022-10-28:
It will be in kernel 6.2 in about 3 months.

closed 2022-10-28 by lwfinger