lwfinger/rtw8852be #41: PCIe Make install error in kernel-5.10

wufenghui, 2023-09-26:


wufenghui, 2023-09-26:
[image]

wufenghui, 2023-09-26:
[image]

lwfinger, 2023-09-26:
I have no idea what is going wrong. I have never seen errors of this sort.

The one thing I have to say is that with an rtw8852be, and a 5.10 kernel, you should be using the rtw89 repo. I provided the one that you are using before rtw89 was ready; however, I no longer maintain the vendor driver.

closed 2023-09-26 by lwfinger

wufenghui, 2023-09-26:
> I have no idea what is going wrong. I have never seen errors of this sort.
> 
> The one thing I have to say is that with an rtw8852be, and a 5.10 kernel, you should be using the rtw89 repo. I provided the one that you are using before rtw89 was ready; however, I no longer maintain the vendor driver.

Can you provide a PCIe driver version that matches the kernel-5.10

lwfinger, 2023-09-26:
Yes, https://github.com/lwfinger/rtw89.git will build on kernels newer than 5.4.