lwfinger/rtw8852be #44: build error in ubuntu24

peacess, 2024-04-27:
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/peace/gopath/src/github.com/lwfinger/rtw8852be/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-31-generic/Makefile:1926: /home/peace/gopath/src/github.com/lwfinger/rtw8852be] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-31-generic'
make: *** [Makefile:635: modules] Error 2

lwfinger, 2024-04-28:
What is the error? It does not show in the listed material.

Change directdory to /home/peace/gopath/src/github.com/lwfinger/rtw8852be, type 'make' and post the output. Then I will be able to help. By the way, with a 6.8 kernel, you should be using the in-kernel driver.

gekasam, 2024-10-12:
And i have this error too
I see what you say about  "by the way, with a 6.8 kernel, you should be using the in-kernel driver." but my wifi work bad (sometimes disconnect, when on another devices all good)  and i try update driver
But now i can`t return wifi
my device is RTL8852BE PCIe 802.11ax 
and Ubuntu 24.04 LTS Linux 6.8.0-45-generic
I try install rtw89  from you and it done, but wifi not connect. 
secure mode disabled

```
remote: Enumerating objects: 9555, done.
remote: Counting objects: 100% (278/278), done.
remote: Compressing objects: 100% (113/113), done.
remote: Total 9555 (delta 191), reused 204 (delta 165), pack-reused 9277 (from 1)
Receiving objects: 100% (9555/9555), 20.99 MiB | 2.96 MiB/s, done.
Resolving deltas: 100% (6160/6160), done.
evgeny@evgeny-work:~$ cd rtw8852be
evgeny@evgeny-work:~/rtw8852be$ make
#rm -f .symvers.8852be
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-45-generic/build M=/home/evgeny/rtw8852be  modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-45-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/evgeny/rtw8852be/platform/platform_linux_pc_pci.o
  CC [M]  /home/evgeny/rtw8852be/os_dep/osdep_service.o
  CC [M]  /home/evgeny/rtw8852be/os_dep/osdep_service_linux.o
/home/evgeny/rtw8852be/os_dep/osdep_service_linux.c:40:6: warning: no previous prototype for ‘_rtw_skb_queue_purge’ [-Wmissing-prototypes]
   40 | void _rtw_skb_queue_purge(struct sk_buff_head *list)
      |      ^~~~~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/osdep_service_linux.c:769:5: warning: no previous prototype for ‘rtw_change_ifname’ [-Wmissing-prototypes]
  769 | int rtw_change_ifname(_adapter *padapter, const char *ifname)
      |     ^~~~~~~~~~~~~~~~~
  CC [M]  /home/evgeny/rtw8852be/os_dep/linux/rtw_cfg.o
  CC [M]  /home/evgeny/rtw8852be/os_dep/linux/os_intfs.o
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:125:14: warning: no previous prototype for ‘rtw_classify8021d’ [-Wmissing-prototypes]
  125 | unsigned int rtw_classify8021d(struct sk_buff *skb)
      |              ^~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:284:5: warning: no previous prototype for ‘rtw_ndev_init’ [-Wmissing-prototypes]
  284 | int rtw_ndev_init(struct net_device *dev)
      |     ^~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:308:6: warning: no previous prototype for ‘rtw_ndev_uninit’ [-Wmissing-prototypes]
  308 | void rtw_ndev_uninit(struct net_device *dev)
      |      ^~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:352:6: warning: no previous prototype for ‘rtw_hook_if_ops’ [-Wmissing-prototypes]
  352 | void rtw_hook_if_ops(struct net_device *ndev)
      |      ^~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:441:5: warning: no previous prototype for ‘rtw_os_ndev_alloc’ [-Wmissing-prototypes]
  441 | int rtw_os_ndev_alloc(_adapter *adapter)
      |     ^~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c: In function ‘rtw_ethtool_get_drvinfo’:
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:511:17: error: implicit declaration of function ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
  511 |                 strlcpy(info->driver, wiphy_dev(wdev->wiphy)->driver->name,
      |                 ^~~~~~~
      |                 strscpy
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c: At top level:
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:626:5: warning: no previous prototype for ‘rtw_os_ndev_register’ [-Wmissing-prototypes]
  626 | int rtw_os_ndev_register(_adapter *adapter, const char *name)
      |     ^~~~~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:747:5: warning: no previous prototype for ‘rtw_os_ndevs_alloc’ [-Wmissing-prototypes]
  747 | int rtw_os_ndevs_alloc(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:800:6: warning: no previous prototype for ‘rtw_os_ndevs_free’ [-Wmissing-prototypes]
  800 | void rtw_os_ndevs_free(struct dvobj_priv *dvobj)
      |      ^~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:910:4: warning: no previous prototype for ‘rtw_init_default_value’ [-Wmissing-prototypes]
  910 | u8 rtw_init_default_value(_adapter *padapter)
      |    ^~~~~~~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:1991:5: warning: no previous prototype for ‘rtw_os_ndevs_register’ [-Wmissing-prototypes]
 1991 | int rtw_os_ndevs_register(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~~~~
[... 7 lines cut ...]
 2430 | int _mi_pm_netdev_open(struct net_device *pnetdev)
      |     ^~~~~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:2452:5: warning: no previous prototype for ‘pm_netdev_open’ [-Wmissing-prototypes]
 2452 | int pm_netdev_open(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:2560:5: warning: no previous prototype for ‘pm_netdev_close’ [-Wmissing-prototypes]
 2560 | int pm_netdev_close(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~~
/home/evgeny/rtw8852be/os_dep/linux/os_intfs.c:3109:5: warning: no previous prototype for ‘rtw_suspend_normal’ [-Wmissing-prototypes]
 3109 | int rtw_suspend_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~
[... cut ...]
 3509 | int rtw_resume_process_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/evgeny/rtw8852be/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-45-generic/Makefile:1925: /home/evgeny/rtw8852be] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-45-generic'
make: *** [Makefile:635: modules] Error 2

```

Depechie, 2025-01-05:
I think I hit the same issue...
Running Proxmox 8.3.2 so Ubuntu 24.04 and getting following details after running `make`

```
Cloning into 'rtw8852be'...
remote: Enumerating objects: 9555, done.
remote: Counting objects: 100% (278/278), done.
remote: Compressing objects: 100% (113/113), done.
remote: Total 9555 (delta 191), reused 204 (delta 165), pack-reused 9277 (from 1)
Receiving objects: 100% (9555/9555), 20.99 MiB | 25.55 MiB/s, done.
Resolving deltas: 100% (6160/6160), done.
root@pve:/home# cd rtw8852be/
root@pve:/home/rtw8852be# make
#rm -f .symvers.8852be
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.12-5-pve/build M=/home/rtw8852be  modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.12-5-pve'
  CC [M]  /home/rtw8852be/platform/platform_linux_pc_pci.o
  CC [M]  /home/rtw8852be/os_dep/osdep_service.o
  CC [M]  /home/rtw8852be/os_dep/osdep_service_linux.o
/home/rtw8852be/os_dep/osdep_service_linux.c:40:6: warning: no previous prototype for ‘_rtw_skb_queue_purge’ [-Wmissing-prototypes]
   40 | void _rtw_skb_queue_purge(struct sk_buff_head *list)
      |      ^~~~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/osdep_service_linux.c:769:5: warning: no previous prototype for ‘rtw_change_ifname’ [-Wmissing-prototypes]
  769 | int rtw_change_ifname(_adapter *padapter, const char *ifname)
      |     ^~~~~~~~~~~~~~~~~
  CC [M]  /home/rtw8852be/os_dep/linux/rtw_cfg.o
  CC [M]  /home/rtw8852be/os_dep/linux/os_intfs.o
/home/rtw8852be/os_dep/linux/os_intfs.c:125:14: warning: no previous prototype for ‘rtw_classify8021d’ [-Wmissing-prototypes]
  125 | unsigned int rtw_classify8021d(struct sk_buff *skb)
      |              ^~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:284:5: warning: no previous prototype for ‘rtw_ndev_init’ [-Wmissing-prototypes]
  284 | int rtw_ndev_init(struct net_device *dev)
      |     ^~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:308:6: warning: no previous prototype for ‘rtw_ndev_uninit’ [-Wmissing-prototypes]
  308 | void rtw_ndev_uninit(struct net_device *dev)
      |      ^~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:352:6: warning: no previous prototype for ‘rtw_hook_if_ops’ [-Wmissing-prototypes]
  352 | void rtw_hook_if_ops(struct net_device *ndev)
      |      ^~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:441:5: warning: no previous prototype for ‘rtw_os_ndev_alloc’ [-Wmissing-prototypes]
  441 | int rtw_os_ndev_alloc(_adapter *adapter)
      |     ^~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c: In function ‘rtw_ethtool_get_drvinfo’:
/home/rtw8852be/os_dep/linux/os_intfs.c:511:17: error: implicit declaration of function ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
  511 |                 strlcpy(info->driver, wiphy_dev(wdev->wiphy)->driver->name,
      |                 ^~~~~~~
      |                 strscpy
/home/rtw8852be/os_dep/linux/os_intfs.c: At top level:
/home/rtw8852be/os_dep/linux/os_intfs.c:626:5: warning: no previous prototype for ‘rtw_os_ndev_register’ [-Wmissing-prototypes]
  626 | int rtw_os_ndev_register(_adapter *adapter, const char *name)
      |     ^~~~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:747:5: warning: no previous prototype for ‘rtw_os_ndevs_alloc’ [-Wmissing-prototypes]
  747 | int rtw_os_ndevs_alloc(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:800:6: warning: no previous prototype for ‘rtw_os_ndevs_free’ [-Wmissing-prototypes]
  800 | void rtw_os_ndevs_free(struct dvobj_priv *dvobj)
      |      ^~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:910:4: warning: no previous prototype for ‘rtw_init_default_value’ [-Wmissing-prototypes]
  910 | u8 rtw_init_default_value(_adapter *padapter)
      |    ^~~~~~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:1991:5: warning: no previous prototype for ‘rtw_os_ndevs_register’ [-Wmissing-prototypes]
 1991 | int rtw_os_ndevs_register(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:2109:6: warning: no previous prototype for ‘netdev_br_init’ [-Wmissing-prototypes]
 2109 | void netdev_br_init(struct net_device *netdev)
      |      ^~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:2384:5: warning: no previous prototype for ‘_pm_netdev_open’ [-Wmissing-prototypes]
 2384 | int _pm_netdev_open(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:2430:5: warning: no previous prototype for ‘_mi_pm_netdev_open’ [-Wmissing-prototypes]
 2430 | int _mi_pm_netdev_open(struct net_device *pnetdev)
      |     ^~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:2452:5: warning: no previous prototype for ‘pm_netdev_open’ [-Wmissing-prototypes]
 2452 | int pm_netdev_open(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:2560:5: warning: no previous prototype for ‘pm_netdev_close’ [-Wmissing-prototypes]
 2560 | int pm_netdev_close(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:3109:5: warning: no previous prototype for ‘rtw_suspend_normal’ [-Wmissing-prototypes]
 3109 | int rtw_suspend_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~
/home/rtw8852be/os_dep/linux/os_intfs.c:3480:6: warning: no previous prototype for ‘rtw_mi_resume_process_normal’ [-Wmissing-prototypes]
 3480 | void rtw_mi_resume_process_normal(_adapter *padapter)
[... cut ...]
/home/rtw8852be/os_dep/linux/os_intfs.c:3509:5: warning: no previous prototype for ‘rtw_resume_process_normal’ [-Wmissing-prototypes]
 3509 | int rtw_resume_process_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/rtw8852be/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.12-5-pve/Makefile:1925: /home/rtw8852be] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.12-5-pve'
make: *** [Makefile:635: modules] Error 2
```